MrChromebox/scripts #1: Write protect blocking all bios options on HP Chromebook 14-ak050nr

jaszhix, 2016-03-28:
Setting boot options, hardware ID, or updating legacy BIOS presents this error:

```Write-protect enabled, non-stock firmware installed, or not running ChromeOS; cannot set boot options.```

@MattDevo I saw you post a [comment](https://www.reddit.com/r/chromeos/comments/49y0hy/boot_from_usb_on_hp_chromebook_14/d0xh0r0) in /r/chromeos saying Bay Trail devices don't need to remove the write protect screws to get the script working - I wonder if this is an exception?

ChromeOS info -

Version 49.0.2623.95 (64-bit)
Platform 7834.60.0 (Official Build) stable-channel kip
Firmware Google_Kip.5216.227.25

I am using developer mode.

Thanks for writing these scripts.

MattDevo, 2016-03-28:
Setting the boot options and hardware ID both require the WP to be
disabled, but updating the legacy boot payload does not. If you're getting
that error for updating the legacy boot, I may need you to do a little
troubleshooting with me to figure out why it's throwing that error.

jaszhix, 2016-03-31:
I rebooted and tried to selected the update legacy bios option first this time, and I got a different error.

Unknown or unsupported device (kip); cannot update Legacy BIOS.

MattDevo, 2016-03-31:
was "kip" in all lower case?  I'm expecting the first letter to be 
capitalized for the string comparison, since all other board names are.  
That's the only reason I can thing that you're getting that error.  But 
I've had other kip owners use the script without issue so I'm slightly 
puzzled

jaszhix, 2016-03-31:
I currently have crouton installed, and a root password set - would that affect anything?

MattDevo, 2016-03-31:
nope, wouldn't affect anything.  my script contains a list of device 
names for each type (eg, baytrail, haswell, etc) so it can determine the 
correct firmware to flash for each device.  Kip is listed in the 
baytrail group, so unless it's a case-sensitivity issue, unsure why it 
wouldn't be correctly identified

jlwilliams3rd, 2016-04-13:
I am having the same issue I have this model hp "HP Chromebook 14-ak050nr 14-Inch Laptop (Intel Celeron, 4 GB RAM, 16 GB SSD)"  When I try the Update Legacy BIOS option i get "Unknown or unsupported device (kip); cannot update Legacy BIOS"

MattDevo, 2016-04-13:
ok, looks like a case-sensitivity issue, I'll try and fix that later today

jlwilliams3rd, 2016-04-14:
Awesome! I'll be waiting :D

MattDevo, 2016-04-14:
should be fixed now

closed 2016-04-14 by MattDevo

jlwilliams3rd, 2016-04-15:
Awesome worked for me!