MrChromebox/scripts #2: Can't install Fedora with Legacy RW

Paviluf, 2016-04-28:
Hello Matt,

I can't install Fedora with Legacy RW on my Falco. The live USB boot to the bootloader, I select to start Fedora and one or two seconds later it reboots. There is an issue opened against Fedora but it's not really active and I don't know if it's a Fedora or firmware problem.

https://bugzilla.redhat.com/show_bug.cgi?id=1135793

Thanks !

MattDevo, 2016-04-28:
@Paviluf have you tried using the 'mem=1952M' parameter?  There's a known issue with some versions of the syslinux bootloader and the stock ChromeOS firmware on Haswell (and Broadwell?) devices.  Depending on how your boot media was created, sometimes writing the iso directly to USB (via dd or Win32DiskImager) can work around the issue.  If you're not going to dual boot / run ChromeOS, then flashing the full firmware will also fix the issue.

Paviluf, 2016-04-28:
I already tried using the 'mem=1952M' parameter but that doesn't work. I wrote the ISO directly to USB with Gnome Disk but that doesn't work too. Do you know if it's something that can be fixed in Fedora or it's really a stock ChromeOS firmware bug ?

I know that flashing the full firmware fix the issue but I'm making some test because, if you remember, my Falco boot sometimes when I plug the power adapter. From what I found that should not happen :
https://productforums.google.com/forum/#!msg/chromebook-central/8zogoqEsMgM/cQ-QVk5pMc0J
https://groups.google.com/forum/#!topic/crouton-central/xo6IU4_d9iM
https://productforums.google.com/forum/#!topic/chromebook-central/Xp8OKGWnmHM;context-place=forum/chromebook-central

MattDevo, 2016-04-28:
I'm not sure whether it's a ChromeOS firmware bug or syslinux one, I just know the issue exists and the 'mem=' workaround.  You might try writing the USB using Rufus if you have access to a Windows machine.

As for your power-on when plugging-in issue, that's likely an issue with the EC (embedded controller) firmware, and not something that can be worked around at the OS level.

Paviluf, 2016-04-28:
It seems that there is a fix in syslinux but it's not merged in Fedora 23 :
https://bugzilla.redhat.com/show_bug.cgi?id=1135793#c27
I just tried Fedora 24 alpha and it's the same. I don't know if it's merged in Fedora 24 alpha.

> As for your power-on when plugging-in issue, that's likely an issue with the EC (embedded controller) firmware, and not something that can be worked around at the OS level.
That's what you said me but it's not 100% sure that an issue with the EC so I make tests, but not at the OS level, rather with the stock firmware (John Lewis Shellball) + Legacy RW (yours and John Lewis), full ROM (CoolStar and John Lewis), etc... ;)

By the way do you know how to update or at least check the version of the EC firmware ?

MattDevo, 2016-04-28:
well, one solution to the issue would be to update the bootloader on the USB install media to the latest testing build of syslinux (6.0.4) which includes the linked fix: https://www.kernel.org/pub/linux/utils/boot/syslinux/Testing/6.04/

Download, extract, and install (or update) to the (mounted) usb install media after writing via dd.

MattDevo, 2016-04-28:
@Paviluf FYI, I verified that updating syslinux to 6.0.4-pre1 fixes the issue.  So download, extract, then:

syslinux-6.0.4-pre1/bios/linux --update /dev/sd[x]

where /dev/sd[x] is the device (not path - eg, /dev/sdb) and you should be good to go (assuming the usb install media is fat32 formatted)

Paviluf, 2016-04-28:
Great thank you very very much ! That's very nice. I will try to have syslinux updated or at least the patch integrated in Fedora 24.

Paviluf, 2016-04-28:
I don't want to abuse but I want to talk to you of 3 others things :D
Can I continue here or you prefer an other way ?

Paviluf, 2016-04-28:
This is what I want to talk about :

- How can I get a valid Falco Hardware ID because I lost mine :(
- There is a [cursor jump bug with Libinput](https://bugs.freedesktop.org/show_bug.cgi?id=94910). If you have a Falco or know someone that does, it can help maybe.
- When I saved VPD and GBB there was this error if I remember correctly:
```
Error accessing high tables, 0x100000 bytes at 0x7f77a000 /dev/mem mmap failed: Resource temporarily unavailable
Failed getting access to coreboot high tables.
```
It don't seem to be a problem but I don't know if my VPD and GBB are valid. I didn't know at the time but the error can be avoided with the "nopat" boot option.
- With the Coolstar or John Lewis full ROM, when I open lid when the Chromebook is completely off, the Chromebook not boot everytime whereas it does with default firmware + Legacy RW
- I can confirm that the Chromebook don't boot when I plug the power adapter with default firmware + Legacy RW so it should be a full ROM bug.

I hope I don't ask too much. If I bother you just say me ;)

MattDevo, 2016-04-28:
for the hardware ID, you can use 'FALCO_C2A-T2Q-D4M' (and set it using the firmware utility script)

the libinput issue is likely an issue with the package itself or configuration, not a firmware issue.  I'd check and see if GalliumOS has the issue, and if so, the devs there will likely be able to help

the GBB matters only for the hardware ID and GBB flags, both of which you can change easily.  The VPD doesn't really matter for a device without ethernet.  So I wouldn't worry about it.

coolstar is probably going to be more receptive to addressing the issue, so I'd ask him on reddit (/r/chultrabook) or IRC (#galliumOS)

closed 2016-04-28 by MattDevo

Paviluf, 2016-04-28:
Thank you Matt for all these infos and for all the things you do and share :)

Paviluf, 2016-04-29:
> the libinput issue is likely an issue with the package itself or configuration, not a firmware issue. I'd check and see if GalliumOS has the issue, and if so, the devs there will likely be able to help

Dudley Du, the author of the Kernel Cyapa driver basically says that my touchpad is broken... I don't think so. Peter Hutterer, the author of libinput, added a patch to discard these jumps ! I tried to contact Hugh from GalliumOS by email, I think he have a Falco, to try to confirm the bug and have a proper fix in the Kernel.

> the GBB matters only for the hardware ID and GBB flags, both of which you can change easily. The VPD doesn't really matter for a device without ethernet. So I wouldn't worry about it.

Ok so GBB don't really need to be flashed if I understand correctly ? And if I still understand correctly the VPD is only used to get back the MAC address ?

> coolstar is probably going to be more receptive to addressing the issue, so I'd ask him on reddit (/r/chultrabook) or IRC (#galliumOS)

I asked him on IRC but unfortunately he don't seem to be very receptive :( That's a big bug nevertheless with, I think, many people concerned. Do you think you can look at it ?

Paviluf, 2016-05-05:
Ok so GBB don't really need to be flashed if I understand correctly ? And if I still understand correctly the VPD is only used to get back the MAC address ?

Do you think you can look at these problems ?

- Unwanted boot when plugged in
- Open lid when the Chromebook is completely off, the Chromebook not boot everytime 

If I bother you just say it ;)

MattDevo, 2016-05-07:
> Do you think you can look at these problems ?
>
> Unwanted boot when plugged in
> Open lid when the Chromebook is completely off, the Chromebook not boot everytime
>
> If I bother you just say it ;)

it's not an issue of being a bother, but my ability to diagnose/fix the issues.  I don't produce any full firmware for Chromebooks (just boxes), so I'd need to clone coolstar's repo with his tweaks and start from there, but wouldn't have any way to test.

Paviluf, 2016-05-07:
> Ok so GBB don't really need to be flashed if I understand correctly ? And if I still understand correctly the VPD is only used to get back the MAC address ?

Can you just say me if it's right ?

Maybe you can work with Coolstar ? John Lewis said me that he knew these problems for 2 years but he don't want to fix them.

MattDevo, 2016-05-08:
> Can you just say me if it's right ?
>
>Maybe you can work with Coolstar ? John Lewis said me that he knew these problems for 2 years but he don't want to fix them.

correct, you don't need to worry about backup up or flashing the GBB and VPD firmware regions.

Given that it's no an issue with the stock firmware, I'd suggest that you just use that instead, with JL's BOOT_STUB update.  Like JL, debugging this sort of issue on hardware I don't have isn't something I'm interested in spending time on, esp when workaround are available

Paviluf, 2016-05-09:
It happen with JL's BOOT_STUB too :( Too bad that nobody is interested in fixing this.

Thank you.