MrChromebox/scripts #3: Integrity of downloaded files

lgierth, 2016-05-06:
I don't feel comfortable executing code without verifying its integrity. @MattDevo I somewhat trust you with compilation of these tools, but I don't trust Dropbox. Could you add verification of SHA256 hashes to the scripts, or alternatively commit them to the git repo?

lgierth, 2016-05-06:
Or just a list of hashes would already be a big help!

MattDevo, 2016-05-06:
The files are all md5 checksummed when built, and verified after
downloading / before flashing

lgierth, 2016-05-06:
Oh I see now, had somehow overlooked that -- The .md5 files don't help me a lot though, because they're downloaded from the same untrusted source. Could you post an md5 list of the dropbox directory? That'd be tremendously helpful. I just don't want my new laptop to be owned right from the start :)

MattDevo, 2016-05-06:
posting a md5 of the directory wouldn't be useful given how often I throw
test builds in there etc.  I'll have to figure something out that's not
overly cumbersome for testing use as well as releases

MattDevo, 2016-07-20:
all files are downloaded from my own domain/server, https://mrchromebox.tech, and like before are md5sum checked after download by the script

closed 2016-07-20 by MattDevo