MrChromebox/scripts #6: Boot and Grub menu gone

Paviluf, 2016-07-04:
Hello Matt,

I just updated the Legacy RW on my Falco and I don't have the "boot menu" (where you can choose to boot from usb) neither the grub menu anymore.

I tested with John Lewis Legacy RW and it's there. I don't know if it's relevant but I flashed the John Lewis default bios too.

I hope that you can fix it !
Thanks !

MattDevo, 2016-07-04:
With my SeaBIOS builds, the prompt for the boot menu only shows if there 
is >1 boot device present; if there's only 1, it boots immediately 
without showing the prompt.  John's builds are completely "stock" and do 
not have this behavior.  If you're not seeing the boot menu prompt with 
a USB device inserted, it could be an issue with the version of SeaBIOS 
I'm using, or it could be an issue with the USB device itself not 
announcing within the USB-spec required time (100ms).  The Grub menu 
showing or not has nothing to do with the legacy boot payload AFAIK.

Paviluf, 2016-07-04:
Sorry I should have said that I'm not seeing the boot menu prompt with a USB device inserted. It worked before with the same usb stick. About the Grub menu, I can see it with John Lewis Legacy RW but not with yours. 

When I boot it seem that the Chromebook stay on the "[warning](http://www.howtogeek.com/wp-content/uploads/2015/02/ximg_54f13e9a6f60f.jpg.pagespeed.gp+jp+jw+pj+js+rj+rp+rw+ri+cp+md.ic.9D9nY51pCC.jpg)" image instead of displaying the boot menu or grub. It disappear only when Fedora start to boot. 

Like you said, maybe there is an issue with the version of SeaBIOS you are using that don't discard the warning screen.

MattDevo, 2016-07-04:
ok, I see the problem, there was a file missing from the build.  Go 
ahead and re-run the script and it should work correctly now

Paviluf, 2016-07-04:
That works now ! That was super fast, thank you so much Matt.
Now, I will install Fedora 24 that can boot perfectly with the live usb, [thanks to you](https://github.com/MattDevo/scripts/issues/2#issuecomment-215345665) that confirmed that the syslinux patch worked and because of this that is now included in Fedora. You are awesome !!!

closed 2016-07-04 by Paviluf