MrChromebox/scripts #7: stumpy firmware link is doa

bi0tech, 2016-07-12:
The source for the stumpy firmware from the sources script is coming up dry.  Not sure if there was an updated version or if it just vanished.

https://dl.dropboxusercontent.com/u/98309225/coreboot-seabios-stumpy-20160623-mattdevo.rom

MattDevo, 2016-07-12:
fixed, filename wasn't corrected at last update

closed 2016-07-12 by MattDevo

bi0tech, 2016-07-12:
What is the correct name for stumpy?  Tried 0704, 0710, 0711 for giggles, but still nada.

MattDevo, 2016-07-12:
I don't recommend flashing directly, but if you're insistent on doing so, just look at the corrected filename in sources.sh.

bi0tech, 2016-07-12:
That's the problem though sources.sh references a file link that doesn't exist.

MattDevo, 2016-07-12:
whoops, forgot to push the commit.  should be fixed now for reals, but plz use the script to flash

bi0tech, 2016-07-12:
Ill give it go.  Wanted to get it put back together before the cat finds where I have all the pieces.

commit 6943a3a4 (2016-07-12): sources: fix filename for stumpy (close #7)