MrChromebox/scripts #13: The script doesn't work anymore

Paviluf, 2016-07-27:
Hello Matt,

The script doesn't work anymore for me. Here is what's happening :
```
$ cd ~; curl -L -O http://mrchromebox.tech/firmware-util.sh; sudo bash firmware-util.sh

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   178  100   178    0     0    761      0 --:--:-- --:--:-- --:--:--   763
100   178  100   178    0     0    213      0 --:--:-- --:--:-- --:--:--   546
100   257    0   257    0     0    260      0 --:--:-- --:--:-- --:--:--   260
100  1007  100  1007    0     0    849      0  0:00:01  0:00:01 --:--:--   849

Getting device/system info...
Failure reading current firmware; cannot proceed.
```
I have a HP Chromebook 14 with shellball from John Lewis as a stock firmware and your Legacy RW.

Thanks !

MattDevo, 2016-07-27:
Under what OS / kernel are you running the script?  Does running it multiple times produce the same result?  That error indicates that reading your current firmware via flashrom didn't succeed, and I recall JL mentioning some kernel config issues with Fedora and kernel 4.5 (IIRC)

Paviluf, 2016-07-27:
I'm on Fedora 24 with Kernel 4.6. Running it multiple times produce the same result. I just looked on the G+ JL posts but didn't found anything related.

MattDevo, 2016-07-27:
can you try using a live USB with a different distro / kernel?  I'm unable to find the post by JL that I was thinking of, but I did find another where he suggested booting with the kernel parameter 'nopat' to correct a certain flashrom error

Paviluf, 2016-07-28:
I had already tried with "nopat",  that I use to avoid a harmless error message, but that doesn't work too. I tried with Ubuntu 16.04.1 with kernel 4.4 I think and that worked. What can break the script on Fedora / Kernel 4.6 ?

MattDevo, 2016-07-28:
> What can break the script on Fedora / Kernel 4.6 ?

it's not an issue with the script, it's an issue a specific kernel config item conflicting with the flashrom binary, which is used to read/write/modify the firmware.  I don't have the time to debug an issue like this that is with a 3rd party app and seems to affect a very limited number of users/configurations

Paviluf, 2016-07-28:
Ok I will report it to Fedora guys. Thanks !

MattDevo, 2016-07-28:
it could be a flashrom issue that just happens to conflict with the Fedora config.  I'd have to install/build/test here to know either way

Paviluf, 2016-07-28:
Should I wait before reporting anything to Fedora since we don't know if it's a Kernel or Flashrom or something else issue ?

I noticed that Flashrom (flashrom-0.9.7) is in the Fedora Repo. Maybe I can test something ?

MattDevo, 2016-07-28:
as with most of the open-source components used with ChromeOS, Google maintains their own repo/branches which are slightly different from the main/upstream version.  Last I checked, upstream flashrom doesn't have the capability to flash the firmware on a running ChromeOS device, so testing with any version Fedora offers wouldn't be useful.

Paviluf, 2016-08-03:
Hello Matt,

Did you found something ?

MattDevo, 2016-08-03:
> Did you found something ?

I haven't had time to test honestly - this is a pretty low priority issue compared to other items I'm working on at the moment

Paviluf, 2016-08-03:
No problem, I will wait :)
If you agree I would like to reopen the issue, just to be sure that won't be forgotten ;)

MattDevo, 2016-08-03:
it's something that would be best addressed by the flashrom developers, but I can ask

vivekjindal, 2016-08-04:
Can you post the output of this command?

zgrep -i devmem /proc/config.gz

vivekjindal, 2016-08-04:
At this stage the script is trying to dump your flashrom to /tmp/bios.bin. However if devmem parameter is set to 'y' then you need recompile your own kernel with option set to 'n'. I would suggest to download chromiumos from arnoldthebat website and run the script from there.

vivekjindal, 2016-08-04:
This is the paremeter name.
CONFIG_STRICT_DEVMEM=???

 │ Symbol: STRICT_DEVMEM [=y]    
  │ Type  : boolean                                               
  │ Prompt: Filter access to /dev/mem                        
  │   Defined at arch/x86/Kconfig.debug:8      
  │     -> Kernel hacking

Paviluf, 2016-08-04:
That doesn't exist "/proc/config.gz"
```
zgrep -i devmem /proc/config.gz
gzip: /proc/config.gz: No such file or directory
```
Thank you for helping !

vivekjindal, 2016-08-04:
The file might be located @/boot with probably same or with some suffix.

vivekjindal, 2016-08-04:
I found this in stackexchange answers:

/boot/config-$(uname -r)

Paviluf, 2016-08-04:
That file "/boot/config-$(uname -r)" is the one. The CONFIG_STRICT_DEVMEM parameter is set to "y" indeed. But, if nothing have changed on flashrom side, that's probably not the problem since Fedora 23 have CONFIG_STRICT_DEVMEM parameter set to "y" too (I just checked) and the script worked. However I can't test anything on Fedora 23 since it have an old syslinux that does not allow it to boot (https://github.com/MattDevo/scripts/issues/2).

vivekjindal, 2016-08-05:
try adding these two parameters to kernel command line:

"nopat iomem=relaxed"

vivekjindal, 2016-08-05:
above information source:

https://www.flashrom.org/FAQ#I_get_.27Can.27t_mmap_memory_using_.2Fdev.2Fmem:_Invalid_argument.27

Paviluf, 2016-08-05:
You fixed it ! Thanks !!! I already used "nopat" but I never heard of "iomem=relaxed" and that's what was needed in complement. For the records, as the link says, and by the way thanks for sharing it, CONFIG_STRICT_DEVMEM is not relevant now.

@MattDevo On my Falco the option "Restore Stock Firmware" is disabled but I can enable it with the "Unlock Disabled Functions". I was wondering why it's disabled and if it's really risky to use it. I already have a stock firmware from John Lewis but, since I prefer your work, I want to install your stock firmware if it's possible. Thanks !

MattDevo, 2016-08-05:
>  I was wondering why it's disabled and if it's really risky to use it. I already have a stock firmware from John Lewis but, since I prefer your work, I want to install your stock firmware if it's possible. Thanks !  

The script will selectively enable/disable functions based on your device, OS, and type of firmware installed -- to help prevent users from selecting an option that's not needed or optimal.  In your case, I'd guess that it disabled that option because it detected that you were running some variant of the stock firmware already, and there was no need to restore to stock.  

In terms of stock firmware, there's likely no difference between the shellball ROMs offered by my script and John's, besides the fact that I pre-set mine with a valid HWID.  Both are simply shellballs extracted from the factory recovery images for a specific device.

Paviluf, 2016-08-05:
Oh ok, so no need to flash again a stock firmware. Unless if there is updates (fix, new feature, etc..) to these stock firmwares. Is it the case ?

MattDevo, 2016-08-05:
the stock firmware changes basically never, no need to update it at all

Paviluf, 2016-08-05:
OK thanks ! I guess I can close the issue now.
Thank you guys, you are awesome !

closed 2016-08-05 by Paviluf