MrChromebox/scripts #15: Full ROM question

Paviluf, 2016-09-16:
Hello Matt,

Nice to see that you now provide your own full rom! If you [remember](https://github.com/MattDevo/scripts/issues/2#issuecomment-215387400) with the Coolstar and John Lewis full rom, the Chromebooks sometimes boot when you plug the power adapter when the Chromebook is completely off and when you open the lid, still when the Chromebook is completely off, the Chromebook not boot everytime. These problem are not present in stock firmware + legacy rom.

Do you know if these problem are in your full rom too ?

Thanks !

MattDevo, 2016-09-16:
still present as far as I know

Paviluf, 2016-09-17:
Ok thanks. If you find a way to fix these problem let us know ;)

MattDevo, 2016-09-19:
shoot me an email at mrchromebox at gmail -- have something I'd like you to test

MattDevo, 2016-09-20:
also, closing this since it's not a script issue, it's a firmware issue.  Firmware issues should be reported at https://github.com/MattDevo/firmware/issues

closed 2016-09-20 by MattDevo

Paviluf, 2016-09-23:
I reopened an issue https://github.com/MattDevo/firmware/issues/15 at the right place to keep track of it.