MrChromebox/scripts #19: acer cb3-111

christ0f, 2016-12-03:
Hey, used your firmware utility script on my acer cb3-111 so I can boot onto my SD card, I managed to get into the select boot device screen and after choosing my SD im stuck "booting from hard disk"
Just wondered if you had any suggestions as to what could be causing the problem. Cheers!

MattDevo, 2016-12-03:
that sounds like an issue with the way the SD card was created.  I recommend using Rufus to create the install media, assuming you're doing it from Windows

christ0f, 2016-12-03:
Ah okay I created it on linux, i'll jump on my windows machine and retry with rufus hopefully that's what the problem is. Thanks for the hasty reply.

MattDevo, 2016-12-03:
linux is fine too, assuming you just dd the ISO/img to the SD card.  Tools like unetbootin just screw things up

christ0f, 2016-12-04:
Yeah seemed to be an SD card problem, I had dd the ISO to the SD but something must have fucked up (laptop froze for a few minutes then resumed) but after using rufus everything ran smoothly. Good work on the utility script and cheers for the help.

closed 2016-12-04 by MattDevo