MrChromebox/scripts #20: ASUS C301SA identified as C202SA

mem-, 2016-12-09:
I tried to install the RW_LEGACY Firmware using the firmware-util.sh script.

There was no error messages. The first shutdown (i didn't reboot from the script) took forever.
I can proceed with regular boot by hitting CTRL-D. When I try legacy boot with CTRL-L the Chromebook hangs. No visual on the screen and I have to use the power button power off and then power on again. CTRL-U still gives the loud beep.
So, CTRL-L in boot hangs the machine and CTRL-D and CTRL-U acts as before (unless hitting CTRL-L before).

The output from firmware-util.sh is attached.

Regards,
// Mem
[image]

MattDevo, 2016-12-09:
What OS are you trying to boot in Legacy Boot mode?  If you didn't install one (via chrx) and don't have a bootable USB inserted, then the system is going to try (and fail) to boot ChromeOS in legagy mode, which will fail / appear like it is hanging.

The C202SA and C302SA are identical from a hardware perspective (and use the same hardware ID), so the ID doesn't matter.  Shutdown would in no way be affected by the script or installing the RW_LEGACY firmware.

mem-, 2016-12-09:
Hi, 

Thanks for your reply and your explanation about the hardware models. 

I tried to boot an USB with Debian on. I can boot my regular laptop from that memory stick. 

Maybe I should try some other dist or setup? 

Regards, 

// Mem 
-- 
Sent from my Android phone with K-9 Mail.

Matt Devo <notifications@github.com> skrev: (9 december 2016 18:03:03 CET)
> What OS are you trying to boot in Legacy Boot mode?  If you didn't
> install one (via chrx) and don't have a bootable USB inserted, then
> the system is going to try (and fail) to boot ChromeOS in [...]

MattDevo, 2016-12-09:
there's no reason it shouldn't boot debian, it's likely an issue with either the USB itself (SeaBIOS is picky about USB devices) or the way in which the ISO was written to USB.

mem-, 2016-12-10:
Hi again, 

I just tested another USB stick with activity LED. During the Chromebook startup (before dev.mode warning), the USB's led was flashing i rappid speed.

While pressing CTRL-U the USB flashed for a second before/during the dev.mode screen is updated. That can be repeted. 

While pressing CTRL-L the USB's led is lighted (no flashing) and the system hangs with the LED still on. 

I should say that that USB doesn't contain any OS, just other data on some ext filesystem. But the USB's led indicates the the Legasy mode should work and I just have to get an OS on that memory stick.

Regards, 
// Mem 

Matt Devo <notifications@github.com> skrev: (9 december 2016 21:45:49 CET)
> there's no reason it shouldn't boot debian, it's likely an issue with
> either the USB itself (SeaBIOS is picky about USB devices) or the way
> in which the ISO was written to USB.

mem-, 2016-12-12:
Hi again,
I just tested another USB stick with activity LED. During the Chromebook
startup (before dev.mode warning), the USB's led was flashing i rappid
speed.

While pressing CTRL-U the USB flashed for a second before/during the
dev.mode screen is updated. That can be repeted.
While pressing CTRL-L the USB's led is lighted (no flashing) and the
system hangs.
I should say that that USB doesn't contain any OS, just other data on
some ext filesystem. But the USB's led indicates the the Legasy mode
should work and I just have to get an OS on that memory stick.

Matt Devo <notifications@github.com> skrev: (9 december 2016 21:45:49 CET)
> there's no reason it shouldn't boot debian, it's likely an issue with
> either the USB itself (SeaBIOS is picky about USB devices) or the way
> in which the ISO was written to USB.

MattDevo, 2016-12-12:
I'd suggest trying to boot with a USB stick that actually contains an OS that's bootable in Legacy BIOS mode

mem-, 2016-12-12:
Hi, 

I totally agree. I just hadn't a bootable USB at hand but would like to update you that things looks promising. 

Regards, 

// Mem
-- 
Sent from my Android phone with K-9 Mail.

Matt Devo <notifications@github.com> skrev: (12 december 2016 09:23:39 CET)
> I'd suggest trying to boot with a USB stick that actually contains an
> OS that's bootable in Legacy BIOS mode

closed 2017-04-02 by MattDevo