MrChromebox/scripts #21: Latest scripts won't run

gotaug, 2017-01-02:
firmware.sh returns:
8: Syntax error: "(" unexpected

firmware-util.sh returns:
33 [[: not found
33 0: not found
33 0: not found
35: source: not found
36: source: not found
37: source: not found
40: prelim_setup: not found
41: [[: not found
44: menu_fwupdate: not found

I've tried running them in bash and from the command line with the same results.

If the error is on my part, please set me straight.  I've searched for solutions and haven't been able to get any different result.

Thanks,
</Aug==---

MattDevo, 2017-01-02:
exact command line used and OS from which being run?

gotaug, 2017-01-02:
I  put everything on a USB stick, opened the terminal (Ctrl+Alt+F3) mounted it and tried running the files (firmware-util.sh and firmware.sh) from the USB drive. That was how I got the errors.

However,  I just found the following command:
cd; curl -L -O http://mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh

It is running now and seems to be working like a charm.  I was gonna let you know about finding that command, and instead I discovered you'd already replied to my issue report.

I've got an Acer C740 and it boots to Shell> now.

How do I enter the bios or get the option to select a boot device such as USB?

I'm impressed by your quick and concerned response.  

Thanks,
</Aug==---

gotaug, 2017-01-02:
Got it!  I wasn't hitting Ctrl + L fast enough!

musiqbonez, 2017-01-20:
Hello Guru's I'm having trouble installing updated script SeaBios on nephews chromebook. I was able to turn off developer mode and I too get the 2 beeps after enable_dev_usb_boot

MattDevo, 2017-01-20:
@musiqbonez what exactly is the issue?  dev_boot_usb has nothing to do with Legacy Boot mode / SeaBIOS.  dev_boot_legacy is automatically set when updating the RW_LEGACY firmware via the script.

musiqbonez, 2017-01-20:
@MattDevo Im getting the two beeps after Ctrl + U meaning there is no SeaBios on this chromebook

MattDevo, 2017-01-20:
@musiqbonez CTRL+L is used for Legacy Boot Mode / SeaBIOS, not CTRL+U.  On Intel Chromebooks, CTRL+U is used to boot ChromeOS/ChromiumOS from USB, nothing else

musiqbonez, 2017-01-20:
my apologies I meant CTRL-L @MattDevo

MattDevo, 2017-01-20:
@musiqbonez what device are you using, and did you already update the RW_LEGAYC firmware via my script?

musiqbonez, 2017-01-20:
@MattDevo i was trying to access RW_LEGACY firmware via your script but not sure fairly new to this. I tried coping it but it doesnt work

musiqbonez, 2017-01-20:
@MattDevo its a Hp Chromebook Memory 4 GB (Ram)
Intel processor screen size 11.6" 16 GB eMMC

MattDevo, 2017-01-20:
@musiqbonez what device/board name does the script report when you run it (it's in parentheses after the device name)?  What do you mean by 'tried copying it'?  The script is run from a terminal shell using the command listed on the website.  From the menu presented, you select option 1 for RW_LEGACY firmware

musiqbonez, 2017-01-20:
I see your script Flash RW_LEGACY with 1185 lines is to be typed in

MattDevo, 2017-01-20:
@musiqbonez I don't understand what you're trying to do.  You don't run any of the scripts directly from this github repository.  You run it using the command listed on my website at https://mrchromebox.tech/#fwscript.

musiqbonez, 2017-01-20:
Not sure if I'm in the right area. It's prompting me for a Login and passwd which chronos followed by sudo bash seems to get me a couple steps but after that I'm lost just graduated school and still not to savy on all this yet sorry

musiqbonez, 2017-01-20:
How do I get to the terminal/shell is that what the picture I sent you is.

MattDevo, 2017-01-20:
@musiqbonez this is very straightforward:
- power on your device
- log in to ChromeOS (guest is fine)
- open link to my Firmware Utility script linked above
- open shell terminal via CTRL+ALT+T, then type 'shell' and press enter
- copy script command from website, paste into shell
- select option 1 for RW_LEGACY firmware update
- reboot, hit CTRL+L for legacy boot mode

I don't know what picture you are referring to, there's nothing embedded in any of your posts

musiqbonez, 2017-01-20:
It's asking for a local host login:

MattDevo, 2017-01-20:
@musiqbonez not if you follow those steps

musiqbonez, 2017-01-20:
Really appreciate you taking the time out of your busy day to assist me. Thanks A Million

musiqbonez, 2017-01-20:
Ok I will try this

musiqbonez, 2017-01-20:
RW_LEGACY firmware successfully installed/updated  awesome

musiqbonez, 2017-01-20:
It looks like I need a 86x windows 10 OS instead 64x OS I was able to see the windows 10 logo after hitting CTRL-L but in gave a error message 3sec later your pc ran into a problem

MattDevo, 2017-01-20:
@musiqbonez you can't install Windows via RW_LEGACY, you need the UEFI Full ROM for that. Again, what device are you using?

ReddestDream, 2017-01-20:
@MattDevo C740, right @musiqbonez?

musiqbonez, 2017-01-20:
Product:	HP Chromebook 11-v010nr ALL
Serial number:	8CG6431ZLN
Product number:

ReddestDream, 2017-01-20:
@musiqbonez This?

http://store.hp.com/us/en/pdp/hp-chromebook---11-v010nr-(energy-star)

This is Braswell, (N30xx). We don't have a Full ROM for that yet, so you won't be able to install Windows on it ATM. We have a Braswell Chromebook we are currently working on, but we don't have anything ready for release yet.

musiqbonez, 2017-01-20:
Ok understandable thanks for the help

closed 2017-01-20 by MattDevo