MrChromebox/scripts #24: Peppy sysfs backlight interface gone after upgrade to 2017/01/23 firmware

jdbean, 2017-01-25:
Running linux (Debian testing) on a haswell Acer C720.

Tried to update my firmware to 2017/01/23 from a crustier build from a while back.

On reboot I noticed that brightness was now set to max and none of my usual backlight control utilities were functioning. Further investigation revealed an empty /sys/class/backlight/ directory.

I saw a few commits about ACPI vs. Intel graphics driver control of backlight but doesn't appear to be working.

ReddestDream, 2017-01-25:
What kernel version is in your Debian Testing?

phelluy, 2017-01-26:
Same issue for me (acer C720P Ubuntu 16.04). After the 2017/01/23 legacy firmware update I cannot use xbacklight anymore. Reverting to software xrandr method and waiting for a firmware fix.

ReddestDream, 2017-01-26:
So, it appears that the backlight control is broken only on the Legacy SeaBIOS firmware.

It is not broken on the UEFI firmware.

Linux works on the UEFI firmware. Only issue is that Debian/Ubuntu-based distros need a correction I can help you with after installation because their installers assume the existence of NVRAM, which we do not have implemented yet.

jdbean, 2017-01-26:
I'm running a 4.9 kernel and also tried booting under a 4.8 kernel with the same result.

Correct, I am running a legacy seabios firmware as I'm trying to preserve an existing Debian installation.

ReddestDream, 2017-01-26:
Best if you can update to the UEFI firmware. rEFInd can help you boot your existing install, but it's going to be a bit mess. Best if you could reinstall.

phelluy, 2017-01-26:
Apparently, it is not very difficult to switch from Legacy to UEFI boot in Ubuntu: 1) create a 200 MB EFI partition; 2) use boot-repair : https://help.ubuntu.com/community/UEFI#Converting_Ubuntu_into_UEFI_mode; 3) flash the UEFI firmware; 4) reboot; 5) fix the UEFI startup.sh (or use rEFInd)
This is my plan, when I will have some time...

jdbean, 2017-01-27:
Tempting alternative and I may do a little looking into switching over to EFI at some point but right now I'm hesitant to risk what seems like a potentially complex undertaking. I've rolled back to a backup firmware and will cross my fingers that the issue will be resolved for the seabios  roms in the future.

Thanks for the info!

coolstar, 2017-01-27:
The SeaBIOS roms are an abomination and were supposed to have been discontinued back in July. We do not test the SeaBIOS roms here, yet Matt still wants to push them anyways. Hence why there's issues like this.

MattDevo, 2017-01-27:
@jdbean since there weren't any real issues with the previous SeaBIOS ROMs (from 11/07), I'm just going to roll back to those.  Trying to maintain two separate sets of firmware for every device we support just isn't tenable, and as the UEFI firmware continues to mature, there is less and less reason to continue development for legacy stuff anyway.

closed 2017-01-27 by MattDevo

jdbean, 2017-01-27:
@MattDevo I can appreciate that the burden is is just too much to handle and I really appreciate all the great work that you and others continue to do on this project.

Maybe if I can figure out the details of safely switching over to an EFI-compatible Debian configuration (converting to GPT, setting up the efi partitions, safely configuring my LUKS encrypted boot setup) at some point I will.