MrChromebox/scripts #26: Legacy-FW Load issue: Asus C310SA-DB04

sethjones, 2017-02-21:
Hello,
I am trying to use your installer script for my asus chromebook.  Everything appears to function normally.  However, trying to install the legacy FW seems to fail.  Everything states that it works correctly.  Yet upon reboot it states that the chromeOs firmware is still present.  

Any recommendations on how to troubleshoot?

Thanks,
Seth

MattDevo, 2017-02-21:
@sethjones sounds like you're experiencing this issue: https://github.com/GalliumOS/galliumos-distro/issues/320

Basically, a ChromeOS update breaks the ability to boot in Legacy Boot mode, so you need to roll back to an older/working version, then re-run the RW_LEGACY firmware update.  Also note that there's no option to replace the ChromeOS firmware for your model currently, all you can do is augment it with a working RW_LEGACY firmware.

closed 2017-02-21 by MattDevo

sethjones, 2017-02-21:
Thanks! I appreciate the fast response. Ill give it a shot.