MrChromebox/scripts #37: Pixel(2013) WIP Status?

BubbatheVTOG, 2017-06-13:
Hi @MattDevo,
I'm curious to know what the status is for the full ROM (UEFI or Legacy) for the Pixel(2013) is. I've been reading on /r/chrultrabook and on your site that it is "in testing". Though, I can't find any details about what works or doesn't work, or even if it works. I don't care about supporting other OS's, I just don't want to wipe my Linux install when the battery is depleted. 

What's the deal? Should I flash it or hold off?

MattDevo, 2017-06-13:
well, there's nothing publicly available to flash ATM, as I'm still working on the EC firmware update, and a few other things.  Now that I have a 2013 Pixel things should move along a bit quicker.

That said, if you have an existing Linux install set up with the stock Legacy Boot firmware, there's no need to flash anything just to work around the battery depletion issue - just use my script to set the firmware boot flags and then no worries about the battery running out and resetting the crossystem flag for legacy boot

BubbatheVTOG, 2017-06-13:
> ...there's no need to flash anything just to work around the battery depletion issue - just use my script to set the firmware boot flags...

WHAT!? That is awesome! Everything I have read said that I needed to flash a full ROM to fix that issue. I guess that I'll have to go run your script on a few other boxes today. :satisfied:

closed 2017-06-13 by BubbatheVTOG