MrChromebox/scripts #40: [Kodi Script] Make it so you can remove the 2nd OS from the menu

MrDiamond123, 2017-07-28:


MattDevo, 2017-07-28:
No idea what you're referring to -- can you provide more detail?

MrDiamond123, 2017-07-28:
In the menu of the Kodi Script add a option to remove Ubuntu/GalliumOS/Kodi and the partition.

MattDevo, 2017-07-28:
The ChromeOS recovery process will reset the partitions on the internal storage to stock, there's no reason for the script to duplicate functionality that already exists

MrDiamond123, 2017-07-28:
Oh...

closed 2017-07-28 by MrDiamond123

MattDevo, 2017-07-28:
http://kodi.wiki/view/Chromebox#Resetting_to_Stock

MrDiamond123, 2017-07-30:
@MattDevo What is the partition does the script use? I am going to format it my self

MattDevo, 2017-07-30:
@MrDiamond123 partitions 6 and 7 (KERN-C and ROOT-C), which are devices /dev/sda6 and /dev/sda7 (or /dev/mmcblk0p6 and /dev/mmcblk0p7 if your device uses eMMC instead of a SSD)

MrDiamond123, 2017-07-31:
k thanks