MrChromebox/scripts #41: ARM support?

SeanFelipe, 2017-08-17:
I happily installed Arch on a HP Chromebook 11 and it works great! Last step is to remove the Google screen at the beginning so it boots straight to Arch without having to control-D.

However when I run your script, it says ARM devices are not supported :(

I'm guessing it's a fair amount of work, but thought I'd at least ask? Thanks!

MattDevo, 2017-08-17:
for setting the GBB flags to auto boot, you can do this from the ChromeOS/ChromiumOS command line.  My script just automates the process, and if not running ChromeOS/ChromiumOS, downloads the required utilities to do so.  Since your device is ARM-based, I would just boot ChromiumOS from USB, then run /usr/share/vboot/bin/set_gbb_flags.sh <value>, where value is the flags settings you want.  If it's CTRL+D with no delay, then setting to 0x01 should suffice.

I wouldn't recommend John's instructions as there's no reason to use flashrom manually when ChromeOS/ChromiumOS already automates the process.

SeanFelipe, 2017-08-17:
Question, how do I boot a chromebook into chromeOS from the usb, without overwriting the linux installation?

I inserted a model-appropriate usb recovery stick, selected control-U on the boot menu, and it went into the recovery. However it nuked my install rather than booting off the usb.

Is there a way to use usb to boot into chromeOS without overwriting? Do I use a recovery stick or a more generic chromeOS build?

MattDevo, 2017-08-17:
CTRL-U from the Developer Mode boot screen boots ChromeOS/ChromiumOS from USB.  Recovery images are exactly that, so not what you want to use.  I'd recommend using one of Arnoldthebat's daily or weekly ChromiumOS images: https://arnoldthebat.co.uk/wordpress/chromium-os/ -- just make sure to get the ARM (CARMOS) version.

MrChromebox, 2018-07-19:
closing this as no plans to add ARM support

closed 2018-07-19 by MrChromebox