MrChromebox/scripts #42: Unable to run utility

Boddypen, 2017-09-18:
When I go to run the firmware-util.sh file that I downloaded, I receive this error. I'm not sure what to do.

```
jaffa@cake:~$ sudo bash ./firmware-util.sh 
Error downloading cbfstool; cannot proceed.

Unable to download cbfstool utility; cannot continue
```

MattDevo, 2017-09-18:
what OS/distro/device?  Sometimes the download of supporting utilities fails when the time/date are wrong and the SSL connections to the server fail as a result

Boddypen, 2017-09-18:
I'm on an Acer Chromebook 14 (edgar) with an RW_LEGACY bios, with the internal storage completely overwritten with GalliumOS.
I tried it again this morning, still the same result. The time and date seems to be exact.