MrChromebox/scripts #44: Default to booting from SD card?

iv597, 2017-10-25:
I currently have my Samsung CB Pro (`CAROLINE`) set up to boot into SeaBIOS by default, but I left ChromeOS on the eMMC for when I want to play around with that. I have Arch Linux installed to a microSD card. I'd like to be able to auto-boot the SD card, but I've hit a few roadblocks.

OOTB, SeaBIOS tries to boot from the eMMC, which is of course not an MBR (or generally BIOS-bootable) drive, and it fails, complaining there is no OS (I'm actually somewhat surprised it doesn't fall through to the SD).

If I hit `ESC` before it tries booting from the eMMC, I can hit `2` to boot the SD as expected.

I modified the scripts in `/tmp` at one point in ChromeOS, and put a `vim bootorder` step in after the bootorder file is downloaded. The three SD card lines at the bottom were moved to the very top, and I flashed the BIOS. These changes worked one time out of at least 5 - oddly, it seemed to "unflash" after the one successful flash + boot, and went back to booting the eMMC by default. Future flashes of this same bootorder tweak did absolutely nothing - the eMMC is always the priority, it seems.

USB stick booting works fine, so I'm guessing the SD reader on `CAROLINE` is actually one of the many PCI lines, I just don't know which, and therefore don't know what to move to the top of the file.

Please let me know if there's any info I can provide that would help to debug or resolve!

MattDevo, 2017-10-26:
need to figure out what the proper entry is to promote, so need to pull the firmware boot log using cbmem.  Download from my site: https://mrchromebox.tech/files/util/cbmem.tar.gz  then extract and run: 'sudo ./cbmem -c > cbmem.txt' (or pipe to pastebinit).  You might need to add a kernel param if mem access too restrictive.  pastebin the log and I'll be able to tell you what the proper entry for the sdcard is (and adjust my script accordingly)

iv597, 2017-10-26:
Thanks for the pointer!

https://gist.github.com/iv597/16ac3a8117335da6cb98d4c7f5611931

MattDevo, 2017-10-26:
@iv597 looks like you took that log after resuming from S3 suspend - need it taken from after a cold boot

iv597, 2017-10-30:
I completely forgot to respond to this. I'll get you a new log this evening after work.

iv597, 2017-11-09:
@MattDevo Sorry for the holdup! https://gist.github.com/iv597/e198ad970bdb2599148541787d5f685e

MattDevo, 2017-11-09:
@iv597 ok, I've adjusted the bootorder file to include the PCI address of the SD card reader, so if you reflash RW_LEGACY and choose USB, it should work now.  If not, grab me the firmware boot log again

iv597, 2017-11-11:
Didn't seem to fix the issue, sadly. [Memory dump here](https://gist.github.com/iv597/c6bf064f99628d81a3a5b0a476390b68)

MattDevo, 2017-11-11:
@iv597 whoops, I'd grabbed the wrong address (mmc vs sd) - try again

iv597, 2017-11-15:
Worked great, thanks a bunch!

closed 2017-11-15 by iv597