MrChromebox/scripts #46: UEFI settings page not saving changes.

abrahimladha, 2017-11-24:
The UEFI ROM is not detecting my BOOTX64.efi file, no matter where I copy it to. I decided to then add a boot option from the UEFI settings page to /boot/efi/EFI/BOOT/BOOTX64.efi and make some other changes like boot order and timeout, but none of these changes persist through reboot, even when I use F10 to save them. Is this by design or am I doing something wrong?

MattDevo, 2017-11-24:
known issue, see the firmware issues tracker:
https://github.com/MattDevo/firmware/issues/10

The firmware boots from /EFI/BOOT/BOOTX64.efi.  It needs to be in that path, on the EFI system partition (ESP) on a GPT-formatted drive.  Assuming you have an EFI bootable OS installed, my FAQ page might be relevant:
https://mrchromebox.tech/#faq

closed 2017-11-24 by MattDevo

abrahimladha, 2017-11-24:
thanks for all your hard work!