MrChromebox/scripts #48: Asus Chromebook C300 seaBIOS Black Screen

noahrubin, 2018-01-09:
Problem:
Black screen when attempting to boot into seaBIOS (event after waiting 15 minutes)

Goal:
Run linux live from already created USB (already tested USB and works on other machines)

Steps I've taken:
1) Enable developer mode
2) Enable dev_boot_usb and dev_boot_legacy
3) Run the firmware-util.sh script with options 1 (RW_LEGACY) and 4->1 (1 second delay and legacy BIOS default)
4) Reboot machine
5) Press CTRL-L on boot splash screen to boot into legacy (tried without pressing CTRL-L, same issue)

Is the next best step to install BOOT_STUB firmware? I'm afraid that will completely remove my ability to boot chromeOS and brick my machine.  Is there debug output I can send to help you understand the issue further?  Would be really great to get this working in the next few days.  Thanks in advance for any help!

MattDevo, 2018-01-10:
hi Noah, I'm not aware of any issues with the C300 (QUAWKS), so your problem is surprising.  Unfortunately, if you're not able to boot via SeaBIOS, there' no way to get any debug data from a failed boot attempt.  

I'd suggest you boot ChromeOS via CTRL+D  and reinstall the RW_LEGACY firmware.  You don't need to set the crossystem flags as the script does that for you when flashing.  You shouldn't use CTRL+L when defaulting to Legacy +1s as doing it too late will break SeaBIOS / prevent it from booting.

noahrubin, 2018-01-10:
Thanks for getting back to me. I reintalled using the script to no avail.  I mistyped in my first comment, I have an Asus C300**SA**.  Does that make a difference?

MattDevo, 2018-01-10:
nope, same device; would only be different if you had a C301 vs C300.  Unfortunately I don't have any good suggestions for you, other than to try the UEFI Full ROM if you don't care about running ChromeOS.  If you do, then perhaps try John Lewis' RW_LEGACY firmware instead, though I can't see that working any differently