MrChromebox/scripts #49: Only 2.7GB memory available

dmagyar, 2018-01-20:
Hi! 

I used your script to flash chromeboot on my Samsung Chromebox Series 3. It has the default 4GB memory. When booting Ubuntu I can only see much less in the system:

```
root@chromebox:~# lshw -short | grep 'System\ Memory'
/0/5                        memory         4GiB System Memory
root@chromebox:~# cat /proc/meminfo | grep -i memtotal
MemTotal:        2766048 kB
root@chromebox:~# 
```

Interestingly it seems that the BIOS is reporting this to the kernel:

```
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.0-109-generic (buildd@lgw01-amd64-038) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.5) ) #132-Ubuntu SMP Tue Jan 9 19:52:39 UTC 2018 (Ub
untu 4.4.0-109.132-generic 4.4.98)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.4.0-109-generic.efi.signed root=UUID=78e1fa34-47f1-4da5-86b0-c1f771ac3796 ro
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x04: 'AVX registers'
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] x86/fpu: Using 'eager' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000001000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000020000000-0x00000000201fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020200000-0x000000003fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040000000-0x00000000401fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040200000-0x00000000acd31fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000acd32000-0x00000000acdb1fff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000acdb2000-0x00000000ace75fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ace76000-0x00000000ace76fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000ace77000-0x00000000ace80fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ace81000-0x00000000ace81fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000ace82000-0x00000000acf15fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000acf16000-0x00000000af9fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000014fdfffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.60 by EDK II
[    0.000000] efi:  ACPI 2.0=0xacf28000  SMBIOS=0xacf16000 
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: SAMSUNG Stumpy/Stumpy, BIOS MrChromebox 07/14/2017
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0xacf16 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: write-back
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   0 base 0AD000000 mask FFF000000 uncachable
[    0.000000]   1 base 0AE000000 mask FFE000000 uncachable
[    0.000000]   2 base 0B0000000 mask FF0000000 uncachable
[    0.000000]   3 base 0C0000000 mask FF0000000 uncachable
[    0.000000]   4 base 0C0000000 mask FE0000000 write-combining
[    0.000000]   5 base 0E0000000 mask FE0000000 uncachable
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000]   8 disabled
[    0.000000]   9 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT  
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] Base memory trampoline at [ffff880000099000] 99000 size 24576
[... 79 lines cut ...]
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 64.
[    0.000000]  RCU restricting CPUs from NR_CPUS=512 to nr_cpu_ids=4.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=64, nr_cpu_ids=4
[    0.000000] NR_IRQS:33024 nr_irqs:456 16
[    0.000000] Console: colour dummy device 80x25
[    0.000000] console [tty0] enabled
[    0.000000] clocksource: hpet: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 133484882848 ns
[    0.000000] hpet clockevent registered
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 2494.219 MHz processor
[    0.000036] Calibrating delay loop (skipped), value calculated using timer frequency.. 4988.43 BogoMIPS (lpj=9976876)
[    0.000043] pid_max: default: 32768 minimum: 301
[    0.000050] ACPI: Core revision 20150930
[    0.002810] ACPI: 2 ACPI AML tables successfully acquired and loaded
[    0.003190] Security Framework initialized
[    0.003194] Yama: becoming mindful.
[    0.003210] AppArmor: AppArmor initialized
...
```

Any tips where should I start? With the legacy ChromeOS on in the linux shell I can see the entire 4GB.

Thanks,
Denes

dmagyar, 2018-01-20:
> this seems to be a bug with the UEFI firmware on Stumpy when the DVI connection isn't used, as it thinks there's no connected display until Linux inits the i915 driver and the displayport gets detected

So I need a monitor or a dummy DVI plug? Is there any way to patch this from software?

This is a headless setup so I don't mind if the video does not work.

MattDevo, 2018-01-20:
a dummy DVI plug would solve the issue, yes

MattDevo, 2018-02-04:
@dmagyar this is now fixed in the current (2018.02.04) release

closed 2018-02-04 by MattDevo