MrChromebox/scripts #50: No USB device available to store firmware backup

winhex, 2018-02-06:
Hi! I have Acer Chromebook 14 (CB3-431) and I'd like to install Full ROM firmware. When I do it,  your script prompts me to create a firmware backup, but  I can't do it, because I always get this error:
`No USB device available to store firmware backup.`
I tried several different flash drives and an external hard drive, but nothing has changed. Why can't I create a firmware backup?
p.s. Now in my chromebook installed GalliumOS 2.1, if this is important.

MattDevo, 2018-02-07:
the script just lists unmounted USB devices which are FAT/FAT32 formatted; you may need to manually unmount the USB drive after inserting it but before running the script

winhex, 2018-02-07:
it's not too obvious, I think)

MattDevo, 2018-02-07:
@winhex not sure what you're referring to exactly

winhex, 2018-02-07:
In my mind this message can be made more informative. For example, indicate that the user must unmount the usb device or something like this

MattDevo, 2018-02-07:
@winhex did that fix the issue for you?  I've been unable to reproduce it myself

mildred, 2018-10-05:
I had a similar issue and it comes from the fact that the script is parsing the output of `fdisk -l`. it needs to be in English. if you have another locale enabled, you need to run the firmware utils like this:

`sudo env LC_ALL=C bash firmware-util.sh`

Also, it would be good to set `export LC_ALL=C` at the beginning of the script

MrChromebox, 2018-10-05:
@mildred just at the top of firmware-util.sh?

mildred, 2018-10-07:
At the top, before anything else, so the environment variable is inherited

MrChromebox, 2018-11-18:
@mildred added, let me know if that fixes the issue for you