MrChromebox/scripts #56: Firmware-util script broken?

PieVo, 2018-07-05:
Hi, just tried to set some bootflags, but I could not run the script anymore:

chronos@localhost ~ $ cd; curl -LO https://mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   185    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  1006  100  1006    0     0   1856      0 --:--:-- --:--:-- --:--:--  1856
./sources.sh: line 2: syntax error near unexpected token `newline'
./sources.sh: line 2: `<?xml version="1.0" encoding="utf-8"?>'
Error downloading cbfstool; cannot proceed.

Unable to download cbfstool utility; cannot continue

MrChromebox, 2018-07-05:
the script isn't broken, your internet is (or @galliumos' hosting is, but I'm not seeing any issues here).

closed 2018-07-05 by MrChromebox

PieVo, 2018-07-05:
You're right, now it is working. The syntax error triggered me.
I did post this issue :) so I guess the hosting of file was temporarily down...

Anyway, thanks for your swift reply and development hours you put into the bootloaders!