MrChromebox/scripts #57: Firmware Checksum download fail

IsaacPaczkowski, 2018-07-09:
While trying to a full ROM install, it ends by saying the firmware could not be downloaded and that it cannot flash. Please Help!

MrChromebox, 2018-07-09:
@IsaacPaczkowski might be helpful if you told me what device you were using, and what the exact error is (a picture works well here).  Also, try again in case it was a simple download error (and be glad the script prevented you from flashing a bad file).

IsaacPaczkowski, 2018-07-09:
Lol, yeah, you're right! Dont want bad firmware. My bad on not including the Info: I am Currently using a Samsung xe500c13 (Celes is the codename) and has an Celeron processor (Braswell).
[image]
[image]
[image]

 I have removed the WP screw and enabled dev mode. (Error and scripts pictured:)

MrChromebox, 2018-07-09:
@IsaacPaczkowski something on your end is breaking the download - the file exists on the server, the checksum is correct, and there's no download error in the server logs

IsaacPaczkowski, 2018-07-09:
I'm sure this is probably it, but, could it be my trying to do this on a business network that uses securities such as Smoothwall, and etc? (Can't believe I didn't think about that...)

closed 2018-08-04 by MrChromebox