MrChromebox/scripts #62: hung on coreboot logo (after rom working for several reboots)

goofydev, 2018-08-04:
Hi, first of all, would like to thank you for the wonderful work you have done, and hope someday I can also contribute to the community in some similar way.

I installed the full UEFI ROM on a Samus, and it was working beautifully for a day (through more than a dozen reboots).  There was some error in the running system (suddenly /var was read-only??), and so I rebooted the system.  Since that time, after power-on the machine hangs on the Coreboot logo (it says at the bottom that memory tested OK).  The same happens with a USB stick (which stick boots ok on a different computer).  I cannot get into the EFI menu by pressing any key.  Is there a way to get into the EFI menu?  Do you think somehow the system is bricked (if so how because I was not doing anything with the firmware).  I have not yet put the write-protect screw back and so I was also wondering if somehow a running system can damage firmware?

Thank you for your attention to this issue.

MrChromebox, 2018-08-04:
based on what you've told me, sounds like the SSD failed, causing /var to go RO, and then hanging the firmware on boot device detection at the splash screen.  Short of doing a hard reset (Refresh+Power) and hoping it's hung vs failed, I don't really have any suggestions as the SSD is non-replaceable

closed 2018-08-04 by MrChromebox

goofydev, 2018-08-04:
Thank you for that quick response.  Looks like bad luck (:-(  One more question, if I may.  Do you have any suggestion as to why it would not boot from the USB device?  Put it another way, is there a way of coaxing the firmware to avoid looking at the SSD and booting from a USB device?

MrChromebox, 2018-08-04:
it's not that it won't boot from USB, it's that it's hanging on boot device detection when querying the internal SSD. There's no way to circumvent that besides a custom firmware which disables the internal SSD

Novex, 2022-05-24:
If anyone else comes across this - I managed to get past the coreboot rabbit and into the EFI shell by desoldering the eMMC chip from the board

A bit of an extreme measure for sure, but it allows me to boot from a USB device