MrChromebox/scripts #65: Cant run Firmware Utility Script on i7 Pixelbook

tbayne, 2018-09-13:
When I try to run it...

cd; curl -LO https://mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   185    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0
100  1006  100  1006    0     0    231      0  0:00:04  0:00:04 --:--:--  9490
/dev/mem: No such file or directory
Unable to determine Chromebox/book model; cannot continue.
                                                                                                                         
It's likely you are using an unsupported ARM-based ChromeOS device,
only Intel-based devices are supported at this time. 
---------------------------
Any thoughts?

MrChromebox, 2018-09-13:
You're running this from ChromeOS? What version? 

The error indicates that flashrom couldn't read the current firmware (via /dev/mem) which would be bizarre since flashrom is shipped with ChromeOS.

tbayne, 2018-09-13:
Latest from Dev Channel:

Version 70.0.3538.16 (Official Build) dev (64-bit)

I've updated then did a couple of restart cycles just to be sure.  Same issue.

MrChromebox, 2018-09-13:
I'm running the same build on my i5/256 Pixelbook, no issues at all, /dev/mem exists as expected.  I'd recommend doing a full reset via Recovery Mode, then retry.  Be sure to not select 'enable debugging features' or anything like that prior to the login screen

tbayne, 2018-09-13:
Ok, will give that a try tomorrow.  Something that I just noticed.  Should I be running the firmware-util.sh from a termina session started from crosh?  Or from the terminal program which is running a penguin instance?

MrChromebox, 2018-09-13:
crosh, then type shell/press enter, then run script command

the crostini terminal is a virtualized environment, has no direct access to memory or the hardware, so no way to use the script under it.

tbayne, 2018-09-13:
Thank you for the education - this explains what I was doing wrong.   RW_LEGACY successfully installed.

MrChromebox, 2018-09-13:
great, I'll add a note about this to the script instructions page; that option didn't exist when I wrote them

closed 2018-09-13 by MrChromebox