MrChromebox/scripts #67: Unable to download Full ROM Firmware

julink17, 2018-09-20:
I have tried to setup an Acer Chromebase24 with UFEI Full ROM, but it just keeps failing when it tries to download the ROM.
[image]

MrChromebox, 2018-09-20:
BUDDY actually isn't supported at the moment for UEFI full ROM firmware, so the error is in showing it as available, not in the download failing. I've built a test ROM for it but still waiting on feedback from a tester before I push it live

closed 2018-09-20 by MrChromebox

julink17, 2018-09-20:
I see, my apologies, I only saw that the Broadwell CPU was capable.  Would it be possible for me test it on my machines as well?  I’m not concern about bricking it.

MrChromebox, 2018-09-21:
No, not your error, issue was/is with the script advertising a function that is not available for your device. Shoot me an email at MrChromebox at Gmail and I'll provide instructions