MrChromebox/scripts #69: ChromeOS dependency?

kousu, 2018-09-21:
I'm playing with a Chromebook due largely thanks to your script, but now I'm considering replacing the ChromeOS installation. https://mrchromebox.tech/#fwscript says 

> an be run from any Linux which has a full Bash shell 

but there's this

https://github.com/MrChromebox/scripts/blob/master/firmware.sh#L15-L17

which looks like it requires ChromeOS. Certainly, most linuxes don't have `crossystem` available. Elsewhere, you use `flashrom`, but that depends on this download

https://github.com/MrChromebox/scripts/blob/f3a2e696888c584d4711311baf44cbfe3305b589/functions.sh#L197

which pulls from https://www.mrchromebox.tech/files/util/ instead of this repo. I'm worried that about how much I can rely on this script. If I mess something up will I be stuck? Could I use your script from a livecd, or would I need to reinstall ChromeOS from Google's sources, or will I be totally bricked? I don't have a spare laptop to test my assumptions with.

MrChromebox, 2018-09-21:
I think you are misreading parts of the script, it can be run from any distro with a bash shell, including Live ISO/USB.  I've tested on Ubuntu and Fedora variants as well as a few others (ie, Arch).

> which looks like it requires ChromeOS. Certainly, most linuxes don't have crossystem available.

that code literally says if the script is being run from ChromeOS, then set the crossystem flag, otherwise it's not executed.  There's no issue there.

> Elsewhere, you use flashrom in the non-ChromeOS case, but that depends on this download 
which pulls from https://www.mrchromebox.tech/files/util/ instead of this repo. I'm worried that about how much I can rely on this script.

so your complaint is that I'm pulling a file from my own website vs my github repo?  All of the binary files are pulled from my website, all of the code from this repo -- nice and clean delineation.

>If I mess something up will I be stuck? Could I use your script from a liveusb, or would I need to reinstall ChromeOS from Google's sources, or will I be totally bricked?

if you can boot a GalliumOS/Ubuntu live USB, you can run the script, and do whatever you need to do.

kousu, 2018-09-21:
Thanks for answering so quick, @MrChromebox.

> that code literally says if the script is being run from ChromeOS, then set the crossystem flag, otherwise it's not executed. There's no issue there.

But that _if_ doesn't come with an _else_; I don't really know what `crossystem` does, but it looks like it's setting firmware variables. In other parts (like I linked above) you are using `flashrom` in the _else_, I'm assuming to accomplish the same thing, but that one part doesn't have an equivalent non-chromeos part. I'm just asking so I can understand what risks I need to watch out for.

Thanks for answering my question. If you say you've tested it on different distros then I will trust that you know what you are doing, since you have worked with this hardware extensively.

kousu, 2018-09-21:
I do think it would be better if all of your dependencies were in the repo and that the entrypoint script was built from and embedded them all, using the trick of [encoding a .tar.gz inside of a script](https://stackoverflow.com/questions/10491704/embed-a-executable-binary-in-a-shell-script#10491738). Then your repo could be forked easily and when you get tired of maintaining it or github starts Doing Bad Things or whatever it won't leave your users trapped. I was worried that, if your host goes down, or you decide to pull the plug on the project, or any number of other things, that we would lose the ability to manipulate our hardware.

MrChromebox, 2018-09-21:
> But that if doesn't come with an else; I don't really know what crossystem does, but it looks like it's setting firmware variables. 

it's setting the ChromeOS Developer Mode variable which enables Legacy Boot Mode (CTRL+L).  If you're not running the script from ChromeOS, then there's no need to set the variable as you're obviously already able to boot in Legacy Boot mode, so no else clause.

> I do think it would be better if all of your dependencies were in the repo and that the entrypoint script was built from and embedded them all, using the trick of encoding a .tar.gz inside of a script. 

honestly, I think many people would find it suspect for a shell script to perform such behavior.  The utilities I'm downloading are standard ChromeOS/coreboot tools that anyone else can build from source, I've just statically compiled them for convenience.  Anyone who wanted to fork or take over my script and has the technical knowledge to do so can handle that (I hope) :)

kousu, 2018-09-21:
>  If you're not running the script from ChromeOS, then there's no need to set the variable as you're obviously already able to boot in Legacy Boot mode, so no else clause. 

That wasn't obvious to me. I'm still not sure it is obvious. "Legacy" sounds like BIOS booting to me --- implying non-legacy is UEFI, which plenty of distros support just fine now. A comment explaining why would be useful.

> honestly, I think many people would find it suspect for a shell script to perform such behavior. 

I agree

> The utilities I'm downloading are standard ChromeOS/coreboot tools that anyone else can build from source

Oh they're from coreboot? I can see now that arch has `flashrom` in repos, but I assumed it was something custom or at least obscure because it was getting downloaded from you. That's okay then.

MrChromebox, 2018-09-21:
> That wasn't obvious to me. I'm still not sure it is obvious. "Legacy" sounds like BIOS booting to me --- implying non-legacy is UEFI, which plenty of distros support just fine now. A comment explaining why would be useful.

the crossystem flag is only needed when booting Legacy Boot Mode from the stock ChromeOS firmware.  If you're running my Full ROM UEFI firmware, then everything ChromeOS related goes out the window as it no longer exists. The UEFI firmware is build from upstream coreboot with Tianocore as the payload, and contains none of the ChromeOS-specific functionality of the stock firmware.  This is all explained on my website in much greater detail than would make sense in a code comment as to why a certain flag is only set in one specific situation

> Oh they're from coreboot? I can see now that arch has flashrom in repos, but I assumed it was something custom or at least obscure because it was getting downloaded from you. That's okay then.

flashrom and gbb_utility are ChromeOS tools; cbfstool is from coreboot.

closed 2018-11-18 by MrChromebox