MrChromebox/scripts #70: Has anyone attempted me_cleaner on full rom image?

iflowfor8hours, 2018-09-26:
Has anyone tried running [me_cleaner](https://github.com/corna/me_cleaner) on the full images you're building for some [intel-based](https://github.com/corna/me_cleaner/wiki/me_cleaner-status) platforms? I have a Pixel 2013 that I would like to get this working on, and perhaps going as far as attempting to port [heads](https://github.com/osresearch/heads) to it.

I've been doing a bit of work on porting coreboot to old and new platforms and appreciate how accessible you're making it for chromeos users to have the freedom to run whatever OS they choose.

dimqua, 2018-09-26:
Isn't it possible to build coreboot image with me_cleaner option enabled?

iflowfor8hours, 2018-09-29:
Ah, I misunderstood how flashing the Link worked. I was under the impression that you would not have to unlock using an external flasher because it has two distinct bios slots. My mistake. Onwards with the rpi and porting project. Thanks again.

MrChromebox, 2018-09-29:
if you're referring to the RW-A and RW-B slots of the stock firmware, none of that applies to the UEFI Full ROM firmware, it uses a single image. But even if not, the ME region is separate from the BIOS region, which is all that's getting updated

MrChromebox, 2018-09-30:
definitely drop me a line and let me know, it might be an option to offer down the road

closed 2018-09-30 by MrChromebox

dimqua, 2018-12-12:
@MrChromebox could you recommend a guide how to flash cleaned ME firmware externally? I have Acer Chromebook 14 cb3​-431 with GNU/Linux.

MrChromebox, 2018-12-12:
@dimqua the guide/instructions for ME cleaner applies to Chromebooks running my Full ROM firmware as well (or stock too, for that matter).  You read the firmware with an external programmer, run ME cleaner on it, then flash it back.  There's not really anything Chromebook specific about it.

dimqua, 2018-12-13:
@MrChromebox, ok, thanks for the info. But what about other "blobs", which are modern Chromebooks have? Does your Full ROM firmware include proprietary CPU microcode, for example?

MrChromebox, 2018-12-13:
my Full ROM firmware includes any blobs needed for the device to function properly.  Depending on device/platform, that will include CPU microcode (which isn't a blob IMO - a blob is executable code), memory init/silicon init (MRC, refcode, and/or FSP), and graphics init (GOP driver).  On certain platforms it may be possible to remove certain parts if one is ok with the resulting tradeoffs/loss of functionality, on others there is simply nothing that can be done to remove any.

MrChromebox, 2018-12-13:
yes, there's nothing special about Chromebooks in that regard.  Pretty much everything in coreboot (blob-wise) is platform, not device, specific