MrChromebox/scripts #72: No menu option to backup firmware without installing a new one

mildred, 2018-10-05:
There is no menu entry to backup the existing firmware without flashing a new one. It should be possible to backup from the main menu.

MrChromebox, 2018-10-05:
I'm conflicted about this.  My reservation stems from the fact that in most cases, users have no need to backup/restore their firmware, and by offering that function, it would simply increase confusion as to when it is needed and potentially leading to users getting themselves into trouble.  

Plus, one can simply run 'flashrom -r <output file>' to back up at any time.

mildred, 2018-10-07:
Then perhaps tell about the flashrom -r command on the docs, and more importantly, clearly state when flashing a new firmware when the backup is happening and when the flashing occurs. It's a bit stressing to agree to flash when you don't have backups yet.

penn5, 2018-10-11:
> I'm conflicted about this. My reservation stems from the fact that in most cases, users have no need to backup/restore their firmware, and by offering that function, it would simply increase confusion as to when it is needed and potentially leading to users getting themselves into trouble.
> 
> Plus, one can simply run 'flashrom -r ' to back up at any time.

IMO, you should add this, because in my case the firmware broke the temperature management. I would love to rollback, but no. The firmware utility script didn't even offer to make a backup for me (maybe because it detected I'm already on a non-stock UEFI)

As to the issue with a risk of people being confused, just add a warning, saying it's not normally needed

Stevie-O, 2019-02-18:
Another reason for someone to want to manually run a backup: testing changes the firmware backup logic that's automatically run just before replacing the stock firmware.

MrChromebox, 2019-02-18:
@Stevie-O not quite sure what you mean. the backup prompt being only for stock firmware is by design