MrChromebox/scripts #73: No option to select firmware version

penn5, 2018-10-11:
Hi,

Thanks for all your hard work (in the 5th issue or what?),

Perhaps you could keep old firmwares available for download, in case of an issue?

Thanks!

MrChromebox, 2018-10-11:
part of the problem is that there is some dependence between the script and firmware.  With the most recent firmware update, I added a feature to the script to re-introduce software write protect, but that required reworking the firmware layout, so it can't be used with older firmware builds.  So I'd have to detect when flashing an older version and disable that feature, which just adds complexity on top of managing multiple firmware versions.  There's several similar issues like this (such as changing from md5 to sha1 checksums) which would require significant effort to offer older builds.

penn5, 2018-10-12:
Or you could just keep hosting old builds, and I can just clone this repo
and run manually...?

MrChromebox, 2018-10-12:
I'd need to rework the directory structure, since hosting multiple versions for 50+ devices in a single directory isn't the best (it's not even the best for a single version at this point).  I'd also need to filter out the older versions which were problematic and wouldn't want people to be able to flash.  It's no small order