MrChromebox/scripts #74: Option to save firmware backup locally

hyshka, 2018-10-29:
I get that you'd want the backup firmware on a USB drive in case of failure but as I went through it I thought it strange to not have an option to save the firmware locally. In my case I wanted to save it locally because I didn't have a USB immediately on-hand and I was planning to copy the firmware online as well.

I can help put together a PR if you think this is a valid use case.

MrChromebox, 2018-10-30:
the main reason I didn't implement that as an option was I didn't want
people backing up locally when running ChromeOS, then flashing the firmware
and not being able to access that backup (because it's encrypted and no
longer accessible).  I'd have no problem accepting a PR for a local backup
option provided it wasn't available when running the stock firmware and/or
when running ChromeOS

hyshka, 2018-10-30:
I see where you're coming from. Since ChromeOS encrypt disks by default it wouldn't be recoverable unless the user had taken a second, independent of the firmware scripts, step to copy that file to online storage. It also makes it especially hazardous as scripts offer the option to reboot after flashing new firmware.

I am going to put a little more thought into this and see if I can come up with a reasonably approach.