MrChromebox/scripts #75: firmware-util.sh not properly detecting write protection state

chilaxan, 2018-12-05:


MrChromebox, 2018-12-05:
software and hardware write protect are two different things, the script reports the latter when run from under ChromeOS

chilaxan, 2018-12-05:
I tested editing the boot options using gbb_utility to shorten the delay and that worked.

MrChromebox, 2018-12-05:
did you previously disable the software WP while the hardware WP was disabled?

chilaxan, 2018-12-05:
yes

chilaxan, 2018-12-05:
using `flashrom -p host --wp-disable` and `flashrom -p ec --wp-disable`

MrChromebox, 2018-12-05:
well that's why, the script is defaulting to the more conservative position, since the hardware WP is enabled, and not expecting users to have manually disabled the software WP.  I suppose I can adjust since newer devices with battery sense WP are more likely to have hardware WP re-enabled

chilaxan, 2018-12-05:
makes sense

MrChromebox, 2018-12-05:
what I'll need to do is report both the HW and software WP state, and add a menu option to toggle the SW wp state

chilaxan, 2018-12-05:
im glad to test any changes

MrChromebox, 2018-12-05:
I have a few APL/KBL devices here I can test on, so should be good, but I'll let you know

chilaxan, 2018-12-05:
ok

closed 2019-08-31 by MrChromebox