MrChromebox/scripts #78: Unable to read current firmware - Samsung Series 3

robisaks, 2019-01-09:
Just did a full restore on my Series 3 through the Google restore process after failing to upgrade Chrubuntu from 14 to 16 (woops). After the restore, I attempted to run the firmware-util.sh and got the error below. Tried running from the CTRL+ALT+T->shell and CTRL+ALT+F2->login, both produced the same error. I realize all I need to do is add 'iomem=relaxed' to your kernel parameters, but I have not been able to find which file to make those changes to. 
 
```
Getting device/system info...
Unable to read current firmware; cannot continue.

Either add 'iomem=relaxed' to your kernel parameters,
or trying running from a Live USB with a more permissive kernel (eg, Ubuntu).

See https://www.flashrom.org/FAQ for more info.

```

MrChromebox, 2019-01-09:
and....fixed

closed 2019-01-09 by MrChromebox

dbro, 2020-03-26:
Today I saw the same message when trying to run "sudo bash firmware-util.sh" on stumpy (Intel i5 version), using both Debian and system rescue CD. Let me know if I can provide more information.

MrChromebox, 2020-03-26:
@dbro did you add the kernel parameter like the script told you to?

dbro, 2020-03-26:
Thanks, I had not done that yet, and that made everything work properly. Sorry to waste your time with this already-answered question!