MrChromebox/scripts #79: Race condition during script startup

Stevie-O, 2019-02-19:
The first thing that `firmware-util.sh` does is download `firmware.sh`, `functions.sh`, and `sources.sh` from the `master` branch.

If someone has the bad luck to be in the middle of running that download while an update is being pushed, they'll run the new version of `sources.sh` with the old version of `firmware.sh` (`functions.sh` is a crapshoot.)

The most robust way to eliminate this problem would be either to:

1. Fetch the commit ID of current master, then fetch the files from that specific commit
2. Put files into a release, then download the latest release file.

I've put together a proof-of-concept that works on my test Linux box (though I can't confirm that it works from `crosh` in an unpatched Chromebook, since I don't have one of those.)

You can see it in action here:
```
curl -LO 'https://raw.githubusercontent.com/Stevie-O/bootstrap-download-test/master/util/update-and-run.sh'
bash update-and-run.sh
```

This code will download the .tar.gz for the latest release, unpack it into a 'bdt-files' subdirectory, and then 
run the main script, which then runs three other scripts.

MrChromebox, 2019-02-19:
while I understand how this could be a problem in theory, I'm not seeing the real-world benefit.  Half the point of the current setup is that I wouldn't have to do releases.

joelpurra, 2026-10-13:
Alternatively, download and unpack relevant scripts from the current `main` commit:

- <https://github.com/MrChromebox/scripts/archive/refs/heads/main.zip>
- <https://github.com/MrChromebox/scripts/archive/refs/heads/main.tar.gz>

Would assume that github properly allows the compressed file download to finish, even if there is a new commit while the user waits. Either way, if the compressed file somehow becomes corrupted it should be detected by default during unpacking.

Does require a decompression utility:

- `*.zip`: `unzip`/`gunzip`/`busybox unzip`/`bsdtar` and several others.
- `*.tar.gz`: `gunzip`/`busybox gunzip`/`bsdtar` and several others.