MrChromebox/scripts #80: firmware-util.sh mixes 'auto update' functionality with actual script functionality

Stevie-O, 2019-02-19:
The `firmware-util.sh` script performs two unrelated tasks:
1. Downloads the latest version of the main script files
2. Sets up the execution environment and runs the main script code

This presents a problem when trying to patch any of the 3 main scripts (firmware, functions, or sources) -- for example, to fix an issue -- as one must also patch the `firmware-util.sh` to *not* overwrite the patched version with the version from the `master` branch.

MrChromebox, 2019-02-19:
I'm not seeing a problem, i see a feature. The whole point is to prevent the use of stale files.  If one feels the need to patch my scripts, then editing firmware-util.sh not to overwrite those changes seems like a trivial burden.  Or am I missing something?

MrChromebox, 2025-01-11:
closing as the script now supports using local sources

closed 2025-01-11 by MrChromebox