MrChromebox/scripts #81: Provide an option to pre-download all files before flashing

Stevie-O, 2019-02-19:
It may seem incredible, but even in this day and age, there are regions without ubiquitous Internet access.  They're generally rural areas.  Rural mountain areas are arguably the worst, because with all that rock in the way, even cellular service is incredibly slow or nonexistent.

This weekend, I was in a situation where I could not get AC power and Internet access simultaneously.  Doing a firmware update on battery power is _not_ my idea of a good time.

(One might ask "what's the point of having a Chromebook when you don't have Internet access?" Well, that's why I'm flashing the firmware to turn it into a regular Linux machine!)

MrChromebox, 2019-02-19:
this adds a lot of complexity / reworking for a really small percentage use case.  Seems like you might be better off just downloading the required files for your device and flashing manually?

CiriousJoker, 2019-11-03:
What about just checking if the file is already downloaded and flashing that instead if the user declines a fresh download?

MrChromebox, 2019-11-03:
I could certainly download the checksum file first, compare, and only download if it fails. But really what's the use case for that? The script would need quite a bit of work to properly cache everything locally, and re-use would only work if flashing multiples of the same device, since the tools downloads are pretty tiny

CiriousJoker, 2019-11-03:
Would it be easy to ask the user once at the start of flashing ("redownload everything?")?
If he says yes, all local files (checksums and images) are deleted.

Then you could just replace every curl call with a custom one that does nothing if the file exists.

The use case is if you're on a train or something and dont have (stable) internet, you can't switch the bootloader unless you do it completely manually.

weirdal3333, 2023-11-14:
Other, older, custom firmware installations are built assuming there is no internet access.
Most of the Wii and 3DS cfw installers have this baked in, as well as several other older games consoles...
I understand that a chromebook is not a game console, but this idea of looking for stored files first before looking on the internet should NEVER die.
What if the website goes down? What if you're in a country with limited internet access? There are a LOT of people that this applies to. Chromebooks are cheap and easy to get in less fortunate areas. Not everyone (for example in the US) can afford anything better than mcdonalds/xfinity free trial wifi.

Having the option for predownloaded files (and checksums) is a good way to keep things easy and open for everyone.

MrChromebox, 2023-11-14:
the logistics of doing this and supporting 250+ devices isn't as easy as you think.