MrChromebox/scripts #83: Flash Chip is in an Unknown State - ASUS CN62

pezmo, 2019-02-22:
I have an ASUS CN62 (fizz).

I ran the command
`curl -L -O https://mrchromebox.tech/setup-kodi.sh && sudo bash setup-kodi.sh`

Then chose 6.
It returned the following error message:

```
VERIFY FAILED at 0x00000086! Expected=0x40, Read=0xc0, failed byte count from 0x00000000-0x00ffffff: 0x2
Your flash chip is in an unknown state.
Get help on IRC at irc.freenode.net (channel #flashrom) or email flashrom@flashrom.org with FAILED: your board name in the subject line!
DO NOT REBOOT OR POWEROFF!
FAILED
--------------------------
An error occurred flashing the Full ROM firmware. DO NOT REBOOT!
Press [Enter] to return to the main menu.
```

I've tried to reach out to the flashrom email, but no response yet. I haven't touched it yet so it's still waiting. What can I do to get it back on track? Please let me know if you need any additional information.

MrChromebox, 2019-02-22:
not that it really matters, but the CN62 (Broadwell) is Guado, the CN65 (Kabylake) is Fizz.  

two bytes off at that location is nothing to be concerned with, since it's in a part of the flash that we didn't actually write to (it's not writable on a live system, normally). Flashrom probably shouldn't bother verifying areas it didn't write, but that's a separate issue. But you're fine to reboot, the error is spurious.

pezmo, 2019-02-22:
My mistake, I meant to type CN65. 

Thank you for clarifying this. That's a relief.

closed 2019-03-22 by MrChromebox