MrChromebox/scripts #84: Request for info (bricked)

krioso, 2019-03-04:


krioso, 2019-03-04:
I've used both the legacy and then full on a Guado and a Panther. Most recently, I was using full rom with Gallium with no issues. THis past weekend I updated the full Rom firmware with Gallium, no issues. Tried to reinstall a new OS via UEFI media  I cannot seem to get either box to boot from the USB. Tried all sorts of UEFI and non-UEFI just in case. There were no errors in the update of the full firmware, everythign went exactly as expected. Wondering if any reports of issues with the most recently updated firmware for these two models.

MrChromebox, 2019-03-04:
I have both Guado and Panther here, and neither has issues booting from USB with the 2019/01/04 firmware.  If your USB devices show up under Boot Manager but don't boot when selected, then the USB simply isn't written in a way that's UEFI bootable.  If the device doesn't show up under Boot Manager, then it's an issue with the USB itself and/or the port being used.

krioso, 2019-03-05:
In my earlier message I didn't indicate that I had my display connected to the DisplayPort on both the Guado and Panther. No video indication on either machine after boot. I decided to try the Panther over HDMI. (I have Guado disconnected from my setup at the moment.) Upon Panther boot on HDMI display I do see, for a brief moment, the full rom boot image, but it drops immediately to the UEFI Shell 2.2, regardless of what keys are pressed immediately on boot (Esc, for example). I have UEFI boot media for a number of Linux OS's (PureOS, Fedora 29, GalliumOS 3 Alpha), tried them all, no apparent recognition of the boot media.

MrChromebox, 2019-03-05:
OK, well simply type exit and hit enter, then from the UEFI settings select Boot Manager, then your USB device.

krioso, 2019-03-05:
I am currently installing Trisquel from what I thought was a previously formatted non-UEFI USB install ISO. The Gallium OS 3 Alpha was recognized as well from the UEFI shell, not directly. Other UEFI formatted install media was not recognized and went back to UEFI shell. Now installing Trisquel - in early install steps, there was an indication that UEFI was noted and would be installed in that mode. I will let the install complete and then see how things boot after that, and whether or not DisplayPort works again. Thanks for the responses to this point.

krioso, 2019-03-05:
After Trisquel install, everything boots as expected ot the newly installed OS. Last step is to recheck DisplayPort, but I anticipate no problems. In the end, I had not expect the UEFI shell without DisplayPort indication, and not sure I am clear on the mysteries of why some of the media I thought were UEFI bootable did not seem to work, while what I thought was formatted legacy mode did. Thanks for responding.

MrChromebox, 2019-03-05:
> In the end, I had not expect the UEFI shell without DisplayPort indication

both the coreboot splash screen and the UEFI shell display correctly here when using the DisplayPort output.  I'm not sure why you're seeing differently, but glad you got things sorted

closed 2019-03-05 by MrChromebox