MrChromebox/scripts #86: UEFI Scripot "sources.sh" not work

Massaguana, 2019-04-26:
Hello,

this "Script" are don´t work...

https://github.com/MrChromebox/scripts/blob/master/sources.sh

the links in there are death

In you tutorial https://wiki.mrchromebox.tech/Unbricking are also many death links

Thanks for fix it

MrChromebox, 2019-04-26:
none of those are full links, they are base URLs onto which additional paths and/or filenames are concatenated.  The directories themselves are not browsable by design.

closed 2019-04-26 by MrChromebox

Massaguana, 2019-04-26:
Okay, and how can i use this Script?

MrChromebox, 2019-04-26:
use in what way? this file is a component of the firmware utility script, you run that (per the Readme)

Massaguana, 2019-04-26:
i found this "Script" out of your unbrick Wiki.... to find the correct UEFI rom file name to flash the Chip

MrChromebox, 2019-04-26:
exactly, the unbricking wiki tells you how to get the correct filename for your device from that file - and even gives an example. it doesn't tell you to run the script.

Massaguana, 2019-04-26:
Okay, then I do not understand the wiki... i Know my Chromebox are an Haswell CPU...

MrChromebox, 2019-04-26:
you need to know the device name for your chromebox.  Look it up https://mrchromebox.tech/#devices.  Then use that name to find the correct firmware file.  For example, if you have an Asus CN60, then the device name is panther, the UEFI file name is coreboot_tiano-panther-mrchromebox_20190104.rom, and so the full URL would be https://www.mrchromebox.tech/files/firmware/full_rom/coreboot_tiano-panther-mrchromebox_20190104.rom.  Make sense?