MrChromebox/scripts #89: Gemini Lake Processor Support

TravisArmstrong, 2019-07-02:
I'm using a new (manufactured Fed 2019) Acer R815TN that includes an Intel Celeron N4100 processor. When loading the firmware script the "CPU Type" is blank and I noticed Gemini Lake not listed on supported platforms. Is there a specific reason for this? Seems kind of like an oldish platform but it's obviously still being used on new products, perhaps due to the recent Intel CPU shortages.

MrChromebox, 2019-07-02:
GeminiLake is the successor to Apollolake, it's the latest Atom-based microarchitecture, and devices are just now coming to market -- which is why there's no support yet. Shoot me an email @gmail and I'll provide you a test build, and if it works I'll push to the main/public script. Be sure to include your device/board name (or HWID)

Git-Ham, 2019-12-23:
Is there any update to this by any chance? I'm toying around with a newer Lenovo 500e (PHASER360) with a Gemini Lake processor and am receiving the following error,

Unknown or unsupported device (phaser): Cannot update RW_LEGACY firmware.

Thank you!

MrChromebox, 2019-12-23:
there is no RW_LEGACY firmware available yet, sorry

StefanoMilani, 2020-05-09:
Hi, is there any update about the GeminiLake support?

MrChromebox, 2020-05-09:
still broken on Google's end

Jaydog7k, 2021-01-18:
How's the future of Gemini Lake look? I bought the HP x360 14b BLOOGUARD instead of 14 SONA smh.

MrChromebox, 2021-01-18:
as dark as ever, unfortunately

animeking9901, 2021-01-26:
I'm open to testing a build for BLOOGUARD if it's possible to make a test build

MrChromebox, 2021-01-26:
test build for what? We know with 100% certainty that RW_LEGACY is broken, I can verify that here on my AMPTON test device. And I have no desire to add UEFI support for the 30+ APL/GLK boards out there just to have people bitch about things (mostly windows) not working

animeking9901, 2021-01-26:
well that sucks,as id be using Linux with UEFI,or RW_LEGACY,i just
wanted to run Linux with maybe wine,which crostini and crouton dont
seem to like too much

MrChromebox, 2021-01-27:
you realize that audio, touchscreen, rotation, etc will all not work, or work far less optimally than under ChromeOS? These devices are just not well supported under mainline Linux distros

animeking9901, 2021-01-27:
Touch screen rotation I can live without but audio good grief let me know
when main line kernel support improves and then I'll reconsider

MrChromebox, 2021-01-27:
I have no idea when that will be. I can't possibly keep track of support for 100+ devices (or even a dozen or so platforms)

msangel, 2021-01-27:
Wine is not needed. Sound and touchscreen we can live without. For terminal server there are not much requirements. Just booting, minimal video, keyboard, maybe mouse and network.

animeking9901, 2021-01-27:
Just because you don't need wine doesn't mean others don't

Thomas-DaxBrin, 2021-05-01:
Hello,

Thanks you for your job mrChromeBox.

I have problem in my legacy mode the bootloader tianoCore don't start After press numeric key.

I would like update or change this to fix bug but my FW version is unknow (Google_bobba.11297.252.0)

Have you more information of this problem ?

MrChromebox, 2021-05-01:
@Thomas-DaxBrin unfortunately I don't support Geminilake Chromebooks at this time

LiamTheProgrammer, 2021-08-29:
I have one of these chromebooks too and I really want to be able to install Linux on it, but I can't yet because this isn't supported. Will support for this be added anytime soon?

MrChromebox, 2021-08-30:
@LiamTheProgrammer nope, no intention of adding GeminiLake UEFI Full ROM support

animeking9901, 2021-08-30:
Ouch. Glad I finally just bought a thin client to mess with instead

uniqueusernamelol, 2021-12-07:
> @LiamTheProgrammer nope, no intention of adding GeminiLake UEFI Full ROM support

have you given up on trying, or is there still any development or progress or just anything trying to get it working for gemini lake?

MrChromebox, 2021-12-07:
there is no impediment to UEFI Full ROM support for GeminiLake devices other than my time and willingness to make it happen. Right now I don't have much of either, since doing so will just unleash more hoards of users wanting to run Windows on them

uniqueusernamelol, 2021-12-07:
> there is no impediment to UEFI Full ROM support for GeminiLake devices other than my time and willingness to make it happen. Right now I don't have much of either, since doing so will just unleash more hoards of users wanting to run Windows on them

is there legacy boot mode now (RW_legacy) or is that still broken

MrChromebox, 2021-12-07:
RW_LEGACY/Alternative bootloader menu on GLK cannot boot SeaBIOS nor Tianocore currently

uniqueusernamelol, 2022-01-14:
> there is no impediment to UEFI Full ROM support for GeminiLake devices other than my time and willingness to make it happen. Right now I don't have much of either, since doing so will just unleash more hoards of users wanting to run Windows on them

how long has it taken you in the past to get uefi full rom on other architectures and what is a potential incentive to get you to develop it since idk anybody else doing this

MrChromebox, 2022-01-14:
> how long has it taken you in the past to get uefi full rom on other architectures and what is a potential incentive to get you to develop it since idk anybody else doing this

oh it's already working, it's literally just a question of do I want to build/distribute ROMs for another ~15 devices, given that they won't run Windows (and that's what most users will want), and the lack of ability to unbrick due to Suzy-Q cables being NLA

animeking9901, 2022-01-15:
Personally I'd just setup some sort of automation that automatically
deletes+bans any issues/users that even say the word windows. Problem solved

MrChromebox, 2022-01-15:
from Github? reddit? my email? my twitter?

uniqueusernamelol, 2022-01-16:
> 

there are genuine reasons to be talking about windows?

oojih, 2022-01-16:
Hi, i have an issu, i'm new to this so dont be rude plz
I followed your tutorial to get the acces to the bios of my hp chromebook x360 but even when i disabled the WP i cannot install RW_Legacy cause i got this error message "Unknown or unsupported device (bloog); cannot update RW_LEGACY firmware." 
I made a bootable usb device wtih kali linux on it but i dont know how could i install it without the bios, i'm trying since many days already, if you have an answer for me or if someone could help me it'll be nice 
(Sorry for the bad english i'm french ^^⁾

animeking9901, 2022-01-16:
Not happening rn,due to whiny windows using bitches so the developer isn't
updating things rn

oojih, 2022-01-16:
damn its a succes for me but the version of kali that i have is an older one 
quite sad but its a first time so i'm still happy to make this by myself ^^

animeking9901, 2022-01-16:
I wonder how hard it would be to port full UEFI myself

MrChromebox, 2022-01-16:
as of right now, I provide one-off unsupported UEFI ROMs for GLK devices upon request via email. Flash at your own risk.

closed 2022-01-16 by MrChromebox

RaineX3333, 2022-01-23:
Hi Sir MrChromebox can i request a uefi roms for my hp chromebook 14a Booglet Gemini lake thank you!

uniqueusernamelol, 2022-01-23:
> Hi Sir MrChromebox can i request a uefi roms for my hp chromebook 14a Booglet Gemini lake thank you!

you are supposed to email him at mr.chromebox@gmail.com, not reply to this thread

hexdump0815, 2022-03-10:
in case someone is only interested to get linux running on its gemini lake chromebook and does not care about the fact, that some hardware is not really working (sound, wifi, deep sleep, ...) it might be worth to have a look at https://github.com/hexdump0815/imagebuilder/tree/main/systems/chromebook_octopus and the debian/ubuntu images linked from there - those are using the native chromebook depthcharge booting to boot the linux kernel/system via ctrl-u - it works quite well with mainline kernels - some more info about how the kernel is built and prepared can be found here: https://github.com/hexdump0815/linux-mainline-x86-64-kernel/blob/main/readme.cbo

best wishes - hexdump

LameLad007-Sudo, 2022-04-08:
So question, I'm using a Dell Chromebook 3100 that aperantly has Legacy Boot Support (when the utility script is ran, it mentions the stock chrome BIOS w/RW_LEGACY), would it be possible to get it working since it mentions this despite it using gemini lake? (It's a 2020 model.)

LameLad007-Sudo, 2022-04-08:
The specific gemini lake one is the N4000 Intel Celeron.

LameLad007-Sudo, 2022-04-08:
[image]
So I have no clue if this may or may not be hopeless, but I don't know if this even is possible, but I got hope it might be.

MrChromebox, 2022-04-08:
@LameLad007-Sudo no, there are no payloads which are functional being used as RW_LGEACY payloads on GeminiLake Chromebooks (other than maybe u-boot).

Shazia34, 2022-06-30:
when will support for it come?

MrChromebox, 2022-06-30:
@Shazia34 when will what come? this issue is closed for a reason

saturnnights, 2022-07-12:
okay, i guess this is as good a place as any.  below is a gist link that contains my coreboot/depthcharge CBMEM buffer, first showing a handoff to a built tianocore payload via altfw autoboot, and second showing handoff to chromeos kernel via dev mode boot.

[https://gist.github.com/saturnnights/70aac966e1f8789ec43ab9001295009c](url)

this is my BLUEBIRD [samsung chromebook 4, related to CASTA, OCTOPUS base board, gemini lake], booting a tianocore payload via RW_LEGACY with stock coreboot/depthcharge [active firmware Google_Casta.11297.242.0].

below are the steps that i took to build the tianocore payload, and how i added it via cbfstool and flashed it.

1. git clone https://review.coreboot.org/coreboot
2. cd coreboot
3. git submodule update --init --checkout
4. make crossgcc
5. make menuconfig
6. select board type of google casta
7. make
8. copy coreboot/build/UEFIPAYLOAD.fd to chromebook
9. flashrom -p host -r [FLASH FILE]
10. cbfstool [FLASH FILE] remove -r RW_LEGACY -n altfw/tianocore
11. cbfstool [FLASH FILE] remove -r RW_LEGACY -n cros_allow_auto_update
12. cbfstool [FLASH FILE] add-payload -r RW_LEGACY -n altfw/tianocore -f UEFIPAYLOAD.fd -c lzma
13. flashrom -p host -i RW_LEGACY -w [FLASH FILE]
14. ensure you can boot legacy/altfw: crossystem dev_boot_altfw=1
15. configure legacy/altfw to boot default: crossystem dev_default_boot=altfw
16. power off
17. power on, wait
18. tianocore, GRUB

coreboot used at time of writing:
commit e3ed9cacaa2b372a3efce95308d15aea3806fbf9
tianocore used at time of writing [i think]:
branch uefipayload_202107

there is still some weirdness.  trying to boot it via CTRL-L and selecting a bootloader doesn't work.  it appears to only work at this time via dev_default_boot being set to altfw and letting it autoboot.  i haven't dug in to why that is.

to make sure that i didn't somehow disrupt something on my BLUEBIRD, i picked up a second BLUEBIRD off of ebay, left firmware write protected, updated only the RW_LEGACY region, configured dev_default_boot for altfw, waited for the timeout, and it launched the tianocore payload.

this appears to be a tianocore payload executing via RW_LEGACY on gemini lake.  tested it UEFI booting an operating system with a finnix live image.

maybe someone else can repro?

MrChromebox, 2022-07-12:
> there is still some weirdness. trying to boot it via CTRL-L and selecting a bootloader doesn't work. it appears to only work at this time via dev_default_boot being set to altfw and letting it autoboot. i haven't dug in to why that is.

I only ever tried via the alternative bootloader menu, not as default via crossystem or GBB Flags. interesting there is a difference, implies that it's the AltFw menu that's breaking things

saturnnights, 2022-07-12:
>interesting there is a difference, implies that it's the AltFw menu that's breaking things

that's what it looks like from my experimentation.  it appears to be possible to boot the different entries in the altfw list directly from the "os verification is off" screen by pressing a number corresponding to the entry index in altfw/list within RW_LEGACY. 

[didn't know that was an option, found that by digging through the vboot_reference source tree]

but if you try to do something that should "leave" the altfw menu, it appears to hang.  that includes the esc key, which should return you to the "os verification is off" screen to perform another action, based on my reading of the code.

so, with this poking around, the following appear to happen:

- if dev_default_boot=altfw, will boot the zeroth index in altfw/list after boot timeout.  the payload that comes with it doesn't appear to work [and isn't guaranteed to, i get it], but building and putting tianocore in does work
- no matter what the default boot option is, either via crossystem, GBB, or FWMP, one can still boot the different entries within the altfw list by using the number keys.  this means that dev_default_boot could be disk or usb, but if you, say, press 0 [after swapping the payload out with one that works], tianocore will load.

so, as long as someone doesn't make use of CTRL-L to try to do altfw/legacy boot, you can still use that feature, and even select different payloads.  if you don't remember what you had loaded, you could press CTRL-L to display the menu, see the number you want, power off, power on, then press that number to boot the corresponding payload.

this is clearly not working "correctly", but it appears the situation might not be as bleak as "FW_LEGACY doesn't work at all".

hexdump0815, 2022-07-20:
@saturnnights - i can confirm that your procedure works - tested it on a blooglet octopus (hp chromebook 14a) booting some own uefi debian image ... this is very cool! ... here are my notes about it (had to add some extra steps to make it fully work): https://github.com/hexdump0815/coreboot-cb-rw-uefi-payload

is this supposed to work only for octopus, only for newer chromebooks, for some chromebooks or for most chromebooks supported by coreboot?

even wifi was working out of the box (which always was very troublesome using chromeos or mainline kernels and kpart depthcharge booting) - sleep does not work (but i think this is a problem on newer chromebooks in general maybe - ?), but at least s2idle suspend works ... sound is most probably a bit more tricky to get working, but i think this is quite normal for newer systems

this is very nice, thanks for sharing this and while i'm here thanks as well to @MrChromebox for all your great work!

best wishes - hexdump

saturnnights, 2022-07-22:
@hexdump0815 to answer your question, i will provide some guesses and speculation, because i have not put in all of the effort to track each branch change to be 100% certain.  sprinkle in some "i think", "i'm not sure", "maybe", etc. throughout my response.

two development trains happened at the same time.  one being the chromeos/chromiumos branches of coreboot, depthcharge, and vboot.  the other being development and production hardware for chromebooks.

i'll write about the software development first.

"originally", some time in the past early enough that figuring out when isn't important to this conversation, the functionality of "legacy boot" was introduced.  you press CTRL-L at the "os verification is off" screen, it checks through CBFS to locate the payload.  loads into ram, jumps to address, execution proceeds in the payload [with the assumption this is a valid payload].

straightforward, but a touch inflexible.  you could only have one payload.

at some point, development began on changing that subsystem from being "single payload" to "multiple payloads, with a menu", and the feature changed from "legacy boot" to "alternative firmware".  you could upload multiple payloads with different names, adjust a file listing to include them, and pick which ones you want.  pressing CTRL-L would bring up a selection menu to pick which one, and if you use "dev_default_boot=altfw", it would boot whatever was in index 0.  the other bit that i don't think has been well documented is that at the "os verification is off" screen, you can directly boot whichever payload is in which slot in the index file listing by pressing the number key at that screen, without pressing CTRL-L.  this is an intentionally written feature as a shortcut, from what i am reading in the source code.  those with stoney ridge and later can confirm.

neat.  but...now let's talk about the other development train.

given my limited sample size of chromebook hardware [absolutely pales in comparison to @MrChromebox and his contacts], i'll guess like this.  this would be the software development overlayed with the hardware releases per generation.

[this is speculation.  the actual demarcation is not important.  the important bit is that there's an inconsistency, and it's in gemini lake generation chromebooks]

* kaby lake/amber lake and earlier - "legacy boot"
* gemini lake - ???
* stoney ridge and later - "alternative firmware"

alright, the ??? part.  at the point that platform released, the development cycle [sprint?  epic?  triumph?] for completely implementing their plans for alternative firmware wasn't done.  [my guess, my speculation].  some parts don't work, some parts are a bit obscure, but if you know what to do, you can boot payloads on gemini lake.

the "verified boot" codepath was stable, so, they snapshot that code train for gemini lake [octopus] and shipped it.  the "legacy boot"/"alternative firmware" feature feels like it was a "we'll do it if we can, we're doing nice things" but is not a prod shipping requirement [in my eyes].  this means that gemini lake got an unfinished implementation of "alternate firmware".

[https://groups.google.com/a/chromium.org/g/chromium-os-dev/c/F7gUL7-IPQw](https://groups.google.com/a/chromium.org/g/chromium-os-dev/c/F7gUL7-IPQw) this may be something to support this theory.  a notice to the chromium-os-dev lists that the alternative firmware bootloader wasn't quite working right, so, boot it manually by number key.  this was november 2018.  in [https://www.chromium.org/chromium-os/developer-information-for-chrome-os-devices/](https://www.chromium.org/chromium-os/developer-information-for-chrome-os-devices/), the earliest gemini lake hardware is february 2019.  it's close.  and still speculation.

to the point of how alternative firmware is used in, well, stoney ridge and later.

1. CTRL-L is supposed to invoke a menu.  from that menu, you're supposed to be able to press a number key to select a payload from the contents of "altfw/list" within the RW_LEGACY region within the CBFS flash layout.
2. at the "os verification is off" screen, i think you're also able to press a number key to select a payload, skipping CTRL-L.
3. and finally, if you have "dev_default_boot=altfw", it will use the payload in the 0 index of "altfw/list".

within gemini lake, from my testing, the only thing that i think is broken is number 1 in the above list.  the menu pops up when you hit CTRL-L, but trying to use the menu doesn't work.

numbers 2 and 3 i have repeatedly tested to work on my limited gemini lake hardware.  maybe other gemini lake devices have different branches, but that doesn't make sense when they're supposed to be stabilized on the same reference platform.  i would guess all of gemini lake [octopus] functions like this.

[very much of this is based upon my reading of code, checking the code branches for gemini lake [octopus], apollo lake [coral], and jasper lake [dedede].  i have talked to no one within google, or the chromium os dev team, or the coreboot team, in any capacity outside of my posting these entries on this issue.  i might be completely wrong.  but i don't think i am.  corrections welcome.]

i hope this helps.

alpernebbi, 2022-09-20:
I looked into this a bit, and it looks like `octopus` boards somewhat fallback to the old `payload` file if there's no `altfw/list` file. I've got a `lick` board, so what I did is mostly:

```bash
curl -L -o lick.rom "https://www.mrchromebox.tech/files/firmware/full_rom/coreboot_tiano-lick-mrchromebox_20220718.rom"
cbfstool lick.rom extract -m x86 -n fallback/payload -f tianocore.elf
cbfstool rw_legacy.cbfs create -m x86 -s 0x1c0000
cbfstool rw_legacy.cbfs add-payload -n payload -f tianocore.elf -c lzma

# On ChromeOS
sudo flashrom -w -i RW_LEGACY:rw_legacy.cbfs
```

It still shows the empty pick-a-bootloader menu when I press CTRL+L, but pressing any number boots into the UEFI flow just fine, no matter when I try it. I booted Debian Installer from it, but didn't test any further.

hexdump0815, 2022-09-20:
@saturnnights - thanks a lot for your long summary (looks like i missed it when you posted it) - all this might make sense

while we are at this: is there any gbb flag value to set
```
crossystem dev_boot_altfw=1
crossystem dev_default_boot=altfw

alpernebbi, 2022-09-20:
> i guess "GBB_FLAG_FORCE_DEV_BOOT_LEGACY" is to force the ctrl-l option - "GBB_FLAG_ENABLE_ALTERNATE_OS" might match dev_boot_altfw=1, but is there anything which would match dev_boot_default=altfw? 

It's `GBB_FLAG_FORCE_DEV_BOOT_LEGACY` and `GBB_FLAG_DEFAULT_DEV_BOOT_LEGACY` according to [this commit renaming "legacy boot" things to "altfw"](https://chromium-review.googlesource.com/c/chromiumos/platform/vboot_reference/+/2641346).

MrChromebox, 2022-09-20:
@hexdump0815 if not running ChromeOS, why are you not using the UEFI Full ROM firmware for your device? RW_LEGACY/Altfw is for dual booting only

alpernebbi, 2022-09-20:
I see the coreboot logo as splash, and press ESC and select the USB from the Boot Menu.

MrChromebox, 2022-09-20:
@alpernebbi quite possible the latest Tianocore payload works around issues with the stock firmware which earlier versions did not. Did you try adding altfw/list and selecting from menu that way?

hexdump0815, 2022-09-20:
@alpernebbi - my idea is to boot uefi via rw payload to have the advantage of uefi (works easily for any kind of linux installers) and rw firmware (no write protect removal required, no suzyqable required, no risk to brick anything, can be easily reverted to chromeos for instance before selling a device) ... i'm running it successfully now this way on a glk bloog device and its very nice - booting looks like: dev-mode-screen -> nice coreboot logo -> logo and escape info -> grub ... it goes directly to grub if installed onto emmc - if running from sd or usb one needs to get into the bios via esc key to select the proper boot device ...

hexdump0815, 2022-09-20:
oh then looks like this is what most probably makes sense in such a scenario:
```
VB2_GBB_FLAG_DEV_SCREEN_SHORT_DELAY
VB2_GBB_FLAG_FORCE_DEV_SWITCH_ON
```

alpernebbi, 2022-09-20:
>  Did you try adding altfw/list and selecting from menu that way?

Yeah, this doesn't work. I can run payloads listed in `altfw/list` by pressing their sequence numbers before I enter the menu, but not after then. Not even after I exit the menu. It's not about the payload, it happens with a half-working U-Boot build as well, and it stops error-beeping for the unavailable entries (when there is no 0-numbered default entry).

MrChromebox, 2022-09-20:
> by pressing their sequence numbers before I enter the menu

wait, what? on what screen? the numbers correspond to the entries in altfw/list, so without that, there's nothing to tie a payload to a number

saturnnights, 2022-09-20:
@MrChromebox i wrote a little bit about how you can trigger the index slots for booting from the "os verification is off" screen

> the other bit that i don't think has been well documented is that at the "os verification is off" screen, you can directly boot whichever payload is in which slot in the index file listing by [...]

the CTRL-L menu is busted.  but selecting/executing the payloads blind works.  or, one could view what's in altfw/list by going into CTRL-L [which won't work], reboot, then use the index number to boot the desired payload.

> so, with this poking around, the following appear to happen:
> 
> * if dev_default_boot=altfw, will boot the zeroth index in altfw/list after boot timeout.  the payload that comes with it doesn't appear to work [and isn't guaranteed to, i get it], but building and putting tianocore in does work
> * no matter what the default boot option is, either via crossystem or GBB, one can still boot the different entries within the altfw list by using the number keys [forgot the important bit.  by using the number keys on the "os verification is off" screen].  this means that dev_default_boot could be disk or usb, but if you, say, press 0 [after swapping the payload out with one that works], tianocore will load.
> 
> so, as long as someone doesn't try to make use of CTRL-L to try to do altfw/legacy boot, you can still use that feature, and even select different payloads. if you don't remember what you had loaded, you could press CTRL-L to display the menu, see the number you want, power off, power on, then press that number to boot the corresponding payload.
> 
> this is clearly not working "correctly", but it appears the situation might not be as bleak as "RW_LEGACY doesn't work at all".

MrChromebox, 2022-09-20:
@saturnnights that absolutely should not work, I'll have to check the depthcharge code and see what's going on

[13 more replies]