MrChromebox/scripts #91: flashrom update for CCD

xaki23, 2019-08-28:
please update the util/flashrom build.
the current build does not seem to have support for "-p raiden_debug_spi", aka "suzyqable" or "cros ccd".

MrChromebox, 2019-08-29:
should it? util/flashrom is only used by my script, which must be run from the host device

xaki23, 2019-08-29:
i can assure you  firmware-util.sh is not the only usecase for your flashrom build.
while i am a user of your script, i also used your flashrom build for usecases outside the script simply because it is known to work and available.

but yes, i guess people with a suzyq are ... not the main target audience.
otoh, we might be more likely to send patches/improvements/reasonablefeedback! 

by now i also had to figure out that raiden_debug_spi is not even a default target for the chromeos flashrom branch, you have to enable it in conf/env/makefile.

(so want to just close it? just go ahead and i wont ask again. ;)

MrChromebox, 2019-08-30:
I can update flashrom to support the Suzy-Q cable, I just need to test thoroughly to ensure no regressions on earlier devices, and no obscure dependencies or lib versions users might not be able to support

closed 2024-02-06 by MrChromebox