MrChromebox/scripts #93: Paste in command not working... SSL_ERROR_SYSCALL

EuphoricPenguin, 2019-09-11:
```
chronos@localhost ~ $ cd; curl -LO https://mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:03 --:--:--     0
curl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to mrchromebox.tech:443 
```
I am not quite sure what the exact issue is. http;// leaves me with blank responses, while the https:// URL leads to the above SSL error.

MrChromebox, 2019-09-11:
what device? what version of ChromeOS? seems like an error one would get with expired certificates due to an outdated OS.

EuphoricPenguin, 2019-09-11:
C720P. Yeah, it just hit it's EOL back in June. Does that mean this is impossible? Was interested in doing this b/c of it's status.

MrChromebox, 2019-09-11:
as long as it's up to date it should work. maybe try with -k in the curl parameters as well?

EuphoricPenguin, 2019-09-11:
Same issue...

EuphoricPenguin, 2019-09-11:
Seems like it could possibly be network related, but navigation to your domain isn't blocked, so I'd assume that wouldn't be the issue. However, my Direct PDANet+ hotspot from my phone creates a different error message than my regular home LAN, which is probably because of PDANet's proxy. Is it possible that I could download the .sh script from this repository and run it from the downloads folder in the crosh shell?

MrChromebox, 2019-09-11:
the firmware-util.sh script is just a bootstrapper, it downloads everything it needs dynamically. So downloading it won't help

EuphoricPenguin, 2019-09-11:
So what do you suggest?

EuphoricPenguin, 2019-09-11:
Dang it. Accidentally closed it.

MrChromebox, 2019-09-11:
well, do a ChromeOS recovery to rule out anything on that end. If issue persists, then I can try to reproduce here on mine

EuphoricPenguin, 2019-09-11:
Ok, I'll try that when I'm back at the house.

mooyoul, 2019-12-04:
Hello There, I am Backend Engineer who started hacking Chromebox. 

I have not been experienced this issue, anyway It looks like SSL Handshake related issue. It usually caused by Bad CA Certificate (e.g. MITM Proxy, Outdated CA certificates in the system) or Cipher negotiation failure (e.g. Outdated Client).

If you are having this problem again, Please attach logs from `openssl s_client -connect mrchromebox.tech:443`. I will help!

EuphoricPenguin, 2019-12-04:
Thank you for your response, but I already have Windows running on my machine from months ago. I think you're totally right, HugesNet is doing some sort of MITM to block UDP VPNs (I tried) and possibly some other specific traffic.

GGWP-grats, 2021-02-26:
@mooyoul Hey! Sorry in advanced for my english. But i have same issue when trying to connect to my docker container's 443th port. I run it with -p 443:443 on debian:buster OS.
Inside container I can get access to https page and logs are:
root@9c665b301804:/var/www# openssl s_client -connect 127.0.0.1:443 
```
CONNECTED(00000003)
Can't use SSL_get_servername
depth=0 C = RU, ST = RF, O = "21SCHOOL, Inc.", CN = 127.0.0.1
verify error:num=18:self signed certificate
verify return:1
depth=0 C = RU, ST = RF, O = "21SCHOOL, Inc.", CN = 127.0.0.1
verify return:1
---
Certificate chain
 0 s:C = RU, ST = RF, O = "21SCHOOL, Inc.", CN = 127.0.0.1
   i:C = RU, ST = RF, O = "21SCHOOL, Inc.", CN = 127.0.0.1
---
Server certificate
-----BEGIN CERTIFICATE-----
MIIDbzCCAlegAwIBAgIUBQqrfGbHiis9JxYOFGLQ/CTRI3YwDQYJKoZIhvcNAQEL
BQAwRzELMAkGA1UEBhMCUlUxCzAJBgNVBAgMAlJGMRcwFQYDVQQKDA4yMVNDSE9P
TCwgSW5jLjESMBAGA1UEAwwJMTI3LjAuMC4xMB4XDTIxMDIyNjE1MzYwOFoXDTIy
MDIyNjE1MzYwOFowRzELMAkGA1UEBhMCUlUxCzAJBgNVBAgMAlJGMRcwFQYDVQQK
DA4yMVNDSE9PTCwgSW5jLjESMBAGA1UEAwwJMTI3LjAuMC4xMIIBIjANBgkqhkiG
9w0BAQEFAAOCAQ8AMIIBCgKCAQEAwWGq2nLR+qxLDPacW68jgclChIBbZ6k4iNC3
wF48GMl2EuHI8VPIqoOGnECXwj1OKJD+aWaGqZ8/I5tz76dJr8gIopcm/alAKEVS
YDZtbzKLjDIqqh6I6YHDI67HNRAR0JLxjB0uLwdJ3hXEUug6r9kFvIZENU7dJaAh
5sCOaXRPoGS1+q/xSsJuSPexh6SDJpaB6e/GQl1OXYZz3P9n2/1EKNhD6klkGFN+
Lm3REJ8R0zoI2nIu+SF+QZ4YlwPYLZRQf+y9gz2+thzFFDNaHRteObOqz6zp2Ntm
t6yI9wZ9EZ/LcSRlfdrDfwjLYYBTjnu35j87uRw3TZ5aMYFxNwIDAQABo1MwUTAd
BgNVHQ4EFgQUn3sYr3/dOygCGF9/NY3UQHGatKUwHwYDVR0jBBgwFoAUn3sYr3/d
OygCGF9/NY3UQHGatKUwDwYDVR0TAQH/BAUwAwEB/zANBgkqhkiG9w0BAQsFAAOC
AQEAql4vt/5TGNHx32pVbjKgKGWprAWoBl2bF8hNmenNS7wg/Xe4UOT7zBg+aP5R
mVpRnvlCHZrDQFlqqnQL+VsKciZ/c1MNF+29TeglBWJ6y5+A0oIjqJd4nfsuOb0N
nsUzBCqG3xnZirUUto8gQ93YVHkBJXIuN+adOVL5CKZ5UU1BTlvYz11wy4QC1nxX
SqNNXYS+I0u9QMhBGUo1uZ5/yWFPRZFVL1HqCxfvzYbSiHM5HuayZph7o8gMfW97
vNWCiB4Fcr5JEGiG1ytK/AIS59NvufQ1a5Q+1HawWiRWTt5XNGsVnG1+5xHZBrrp
Emh/FpsIMs8cHITWvuhr8Kwdfg==
-----END CERTIFICATE-----
subject=C = RU, ST = RF, O = "21SCHOOL, Inc.", CN = 127.0.0.1

issuer=C = RU, ST = RF, O = "21SCHOOL, Inc.", CN = 127.0.0.1

---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: RSA-PSS
Server Temp Key: X25519, 253 bits
---
SSL handshake has read 1524 bytes and written 376 bytes
Verification error: self signed certificate
---
New, TLSv1.2, Cipher is ECDHE-RSA-AES256-GCM-SHA384
Server public key is 2048 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : ECDHE-RSA-AES256-GCM-SHA384
    Session-ID: 9BC1CB920FFAF3351920AEBB00A7762FF9549B6BBC5B6C426FED59309BFCB62A
    Session-ID-ctx: 
    Master-Key: A4E00CFD636BF32ED29502C127959A3A5D6F04C41B70943ABD34C3441C45FA591C9D2D11E6C0FA912C902CF08D59FA8A
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 300 (seconds)
    TLS session ticket:
    0000 - 16 77 b4 cb a9 65 11 4a-81 55 19 55 15 e3 d6 b9   .w...e.J.U.U....
    0010 - d7 70 08 88 af af b2 a4-05 b0 db 69 7f 66 fe 03   .p.........i.f..
    0020 - 41 14 4e dc e0 5b 7c 24-a7 24 cf f4 95 8a 46 42   A.N..[|$.$....FB
    0030 - 42 ef 09 9b 49 db 77 3d-fe 1c 28 ca 03 e2 f0 58   B...I.w=..(....X
    0040 - 4a 71 d1 6c 14 4b 7d e3-cd 94 ad 8a 12 b1 58 81   Jq.l.K}.......X.
    0050 - 08 03 c0 3f 29 0e 7b 6a-84 62 47 37 ab e4 12 9b   ...?).{j.bG7....
    0060 - 3c 2d 4e 9b 1c 30 74 c6-6f 1a 2a 06 ba 54 af 74   <-N..0t.o.*..T.t
    0070 - b1 9f 3d be 05 fd 91 41-93 01 c1 1a 86 21 9a 2b   ..=....A.....!.+
    0080 - c0 c4 4e 6d de f9 77 40-a0 26 7d 09 a3 d6 28 01   ..Nm..w@.&}...(.
    0090 - f2 e8 43 75 06 ab c5 5a-4a c4 5e c3 b5 2c 16 43   ..Cu...ZJ.^..,.C
    00a0 - b9 0f 57 7f 81 b5 dc d3-b2 02 98 e4 e0 92 16 8b   ..W.............

    Start Time: 1614354461
    Timeout   : 7200 (sec)
    Verify return code: 18 (self signed certificate)
    Extended master secret: yes
---
DONE
```
But via my original host curl gets `SSL_ERROR_SYSCALL in connection to localhost:443 `
AND ```
openssl s_client -connect 127.0.0.1:443        
CONNECTED(00000003)
write:errno=104
---
no peer certificate available
---
No client certificate CA names sent
---
SSL handshake has read 0 bytes and written 283 bytes
Verification: OK
---
New, (NONE), Cipher is (NONE)
Secure Renegotiation IS NOT supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
Early data was not sent
Verify return code: 0 (ok)
---
```

GGWP-grats, 2021-02-26:
I will be really happy if you could help me!

MrChromebox, 2021-02-26:
not sure what you are trying to do, but you can't easily run my script from a docker container. it needs direct access to the hardware to read/write the firmware

GGWP-grats, 2021-03-07:
sorry, I just missunderstood this issue. :cactus:

closed 2023-10-14 by MrChromebox