MrChromebox/scripts #95: chromebook pixel 2013,manjaro install fail,efibootmgr failed to register the boot entry

whitestarlau, 2019-10-14:
I have a chromebook pixel 2013(LINK),and use the MrChromebox script to flash the (UEFI) Full ROM.
Then I try install the manjaro-gnome,I got the mistake from the installer:

> Command 'grub-install --targer=x86_64-efi --efi-directory=/boot/efi --bootloader-id=Manjaro -- force'returned non-zero exist status 1.
> Installing fore x86_64-efi platform. Could not prepare Book variable:No space left on device grub-insatll: efibootmgr failed to register the boot entry :Input/output error.

Before manjaro,I have already try galliumOS,the installation was successful,but the xface is not good at HIDPI.So I want try the manjaro.

Can anyone help me?

whitestarlau, 2019-10-14:
Sorry for the  spelling mistakes.

> Command 'grub-install --targer=x86_64-efi --efi-directory=/boot/efi --bootloader-id=Manjaro -- force'returned non-zero exist status 1.
> Installing for x86_64-efi platform. Could not prepare Boot variable:No space left on device grub-insatll: efibootmgr failed to register the boot entry :Input/output error.

MrChromebox, 2019-10-14:
try using the clear NVRAM option of my script

whitestarlau, 2019-10-14:
> 
> 
> try using the clear NVRAM option of my script

Which script,please? firmware.sh?or firmware-util.sh? I cant find any guide from the readme.md or your website.

MrChromebox, 2019-10-14:
there's only one: https://mrchromebox.tech/#fwscript

whitestarlau, 2019-10-14:
> 
> 
> there's only one: https://mrchromebox.tech/#fwscript

I'm confused.The script only have 9 choices and this page does not mention NVRAM at all.I try Restore the Stock Firmware,and the Install/Update a full custom firmware image (Full ROM) again. The problem remains unsolved.

whitestarlau, 2019-10-14:
> 
> 
> > there's only one: https://mrchromebox.tech/#fwscript
> 
> I'm confused.The script only have 9 choices and this page does not mention NVRAM at all.I try Restore the Stock Firmware,and the Install/Update a full custom firmware image (Full ROM) again. The problem remains unsolved.

Now I tried to install galliumOS again.Like installing ubuntu, it failed. I found the function clear_nvram() in firmware.sh,but  I'm not professional with Shell Scripting languages,I dont know how to call it.
Please, can you describe it in more detail?

whitestarlau, 2019-10-14:
> 
> 
> > there's only one: https://mrchromebox.tech/#fwscript
> 
> I'm confused.The script only have 9 choices and this page does not mention NVRAM at all.I try Restore the Stock Firmware,and the Install/Update a full custom firmware image (Full ROM) again. The problem remains unsolved.

Finally I got it.There has a "c" input choice but dont  introducing on the Web page.

MrChromebox, 2019-10-14:
> I try Restore the Stock Firmware,and the Install/Update a full custom firmware image (Full ROM) again. The problem remains unsolved.

why would you do that when that's not what I told you to do?

> Finally I got it.There has a "c" input choice but dont introducing on the Web page.

I've updated the documentation and image on my site

closed 2019-10-14 by MrChromebox