MrChromebox/scripts #97: Careena RW_LEGACY not starting TianoCore

adiabat, 2019-11-07:
Hi - if this is the wrong place to put issues, please let me know the right place and I'll report there.

I got a new CAREENA chromebook, never booted or logged in and entered developer mode.  There's an "enable debugging features" link on the initial welcome screen which I haven't seen on other chromebooks, and I clicked that and entered a root password.  Then ctrl-alt-f2'd into a shell with that newly entered password and got the mrchromebox script to work, and installed the RW_LEGACY firmware; everything looked OK.  The shell login mentioned `enable_dev_usb_boot` so I ran that and it said "success".

After rebooting however, it doesn't recognize USB sticks or SD cards that other chromebooks are OK with for booting; CTRL-U just beeps.  CTRL-L asks if I want 1. U-Boot or 2. TianoCore, however after that text is displayed, no keyboard input has any noticeable effect and I need to hold the power button 8 seconds to turn the whole thing off.

If there's some setting or something else to try, please let me know.  I know the AMD chipsets are new / not really supported but am very willing to try whatever / help poke around if there are promising directions on this chipset.

Also- thanks for all your work on this stuff!  I've been running real linux on cheapo chromebooks for 4 or 5 years :)

MrChromebox, 2019-11-07:
> Hi - if this is the wrong place to put issues, please let me know the right place and I'll report there.

firmware-related issues go under the firmware issue tracker :)

> There's an "enable debugging features"

don't touch this, only makes for problems. Unless you're building/debugging ChromeOS/ChromiumOS.  If it were needed my instructions would tell you to enable it.

> The shell login mentioned enable_dev_usb_boot so I ran that and it said "success".

not relevant, only used to boot ChromeOS/ChromiumOS from USB

> CTRL-U just beeps. 

expected, you don't have a USB w/ChromeOS inserted

> CTRL-L asks if I want 1. U-Boot or 2. TianoCore, however after that text is displayed, no keyboard input has any noticeable effect and I need to hold the power button 8 seconds to turn the whole thing off.

some AMD Chromebooks have issues with legacy/alt-OS mode, CAREENA appears to be one of them.  I have 3 here, 2 work and 1 freezes like you are experiencing. Issue is on Google's end unfortunately.

closed 2019-11-07 by MrChromebox

adiabat, 2019-11-07:
OK thanks, will keep an eye out for any developments.

I'd offer to test any new script / firmware on mine (sounds like you have a bunch of them though!)

adiabat, 2019-11-08:
Not sure if I should open an issue over on the firmware repo or if this is not useful, but --

I went back to chromeos developer mode (CTRL-D) and logged in, gave it a network connection and chromeos did some updates.  After updates happened, re-ran the mrchromebox script for RW_LEGACY, and now with CTRL-L I can get into TianoCore.  (The 1 / 2 options aren't displayed; just a white screen with "Alternative Firmware Menu" in the center and no options, but pressing 2 does get to the rabbit logo of TianoCore).  

I've tried Debian & Lubuntu installers and they seem to choke on display issues (I get a kind of interlaced / mangled text mode that's still readable enough to see Xorg display errors).  Even with the mangled video, I managed to get into a terminal and run aplay and got sound to come out from the headphone jack (not speakers though).  Suspend also seemed to have the same TPM issue that Apollo Lake chromebooks have where it wipes the legacy boot functionality on resume.

Anyway just reporting the change; running google's chromeos update may also allow the CAREENA laptops which freeze to enter legacy boot.  I'm happy to open an issue at the other report & report what works / doesn't work with various things I try if that would be helpful.  If not, no problem.

MrChromebox, 2019-11-08:
on the AMD devices I've tested here, Ubuntu (19.04+) boots without issue, but internal storage is not usable/bootable. The suspend/resume issue is same as Kabylake etc -- need latest kernel with patch to support CR50 TPM, or disable suspend until merged into released kernel. Haven't tried Debian or Lubuntu so can't speak to those

adiabat, 2019-11-08:
Interesting -- will try Ubuntu 19.04.  With Lubuntu 19.04 (but not Lubuntu 19.10!) I am able to see the eMMC partitions and read the chromeos filesystem, but I haven't tried repartitioning it. (TianoCore can't see it so guess wouldn't be able to boot anyway)

MrChromebox, 2019-11-08:
> (TianoCore can't see it so guess wouldn't be able to boot anyway)

possible to boot grub on USB/SD, and have root/boot partitions on eMMC. But untested :)