MrChromebox/scripts #98: Falco not showing firmware menu nor grub menu

malomehi, 2019-11-28:
Hi,

I have updated my HP Chromebook 14 (Falco) to the latest version of the full firmware `20191125`.
Before the update I was running the previous version `20191028` which worked perfectly fine.
My system runs Debian Bullseye.

After the update the firmware screen (black screen with white coreboot rabbit) is not shown anymore.
The Debian grub menu is not shown neither.
But if I wait a few seconds Debian boots normally.

If I press any key before Debian starts booting it remains on black screen forever and I have to shut down the system.

I think grub somehow works because if I increase the value of `GRUB_TIMEOUT` in the grub config file then it takes longer to start booting Debian.
But if I press any key before boot starts then nothing happens, even if I press enter which should just start booting Debian with the default options.

I cannot re-install the firmware from Debian because I get this:
```
Unable to read current firmware; cannot continue.

Either add 'iomem=relaxed' to your kernel parameters,
or trying running from a Live USB with a more permissive kernel (eg, Ubuntu).
```
I used to update the firmware using a Ubuntu live usb stick. But I cannot boot from a usb stick now as I cannot enter the firmware menu to do so.

Any suggestion on what to do to fix this issue?

Thanks.

malomehi, 2019-11-28:
Update:
Before Debian starts booting it literally looks like the screen is off as there is no light on the screen at all, the screen looks pure black, totally off.
It seems that Debian switches the screen on once it starts booting.

MrChromebox, 2019-11-28:
firstly, follow the instructions to add the required kernel param to your grub boot args.  Then reboot, and I'll have you pull the firmware boot log so I can see what's going on / why the backlight isn't coming on

malomehi, 2019-11-28:
Thanks for the quick response!
I have updated the kernel param which seems to work (I should have tried that in the past instead of booting with a usb stick every time I updated the firmware...)

MrChromebox, 2019-11-28:
ok, great. now if you can pull the firmware boot log for me and throw it on pastebin (eg) that would be great.  
Download https://mrchromebox.tech/files/util/cbmem.tar.gz and extract, then run: `sudo ./cbmem -1 > cbmem.log`

malomehi, 2019-11-28:
Ok, I have pulled the logs. Pasted here:

https://pastebin.com/uaAbSsKf

I hope that helps. Thanks!

MrChromebox, 2019-11-28:
hmm, looks like the display was correctly initialized at 1366x768, so a bit baffling. Will likely need to have you flash a debug build and try to see what's going

malomehi, 2019-11-28:
That sounds good. Should I try re-flashing the firmware using `firmware-util.sh` as always to see whether that makes any difference?
Also, is there any way to go back to the previous version in case we cannot find what is causing this issue?

MrChromebox, 2019-11-29:
no, reflashing won't do anything useful. I'll provide you a different link to flash a test/debug build.

If we can't figure it out quickly, I'll revert to previous display init and post an updated build

MrChromebox, 2019-11-29:
@malomehi  shoot me an email (MrChromebox at gmail) and I'll provide you with a link to a test/debug build

malomehi, 2019-11-29:
Fantastic!!
Your test build works. I can access the firmware boot settings and grub is correctly displayed too.
The system also boots normally.

Newgreenbird, 2019-12-04:
@MrChromebox  hi， I had the same problem。 my pixelbook2017（Beta version）Hope to get your help。

MrChromebox, 2019-12-04:
@Newgreenbird what exactly do you mean? there are no similar problems with the Pixelbook, and not sure what you mean by 'beta version.' Please start another topic as the issue with FALCO is unrelated to anything you might be experiencing on EVE

closed 2019-12-04 by MrChromebox

Newgreenbird, 2020-01-10:
sorry，My equipment has been sold。测试版，据卖家说是实验室版本的机器，键盘和开关，与正式版有明显区别。cpu-M7 不是i7。chrome os 没有声音，和触摸板驱动。非常感谢您的 回复！我最近购买了三星chromebook pro m7-6y75 16G ram  64G rom   用您的bios，刷入win10。  cpu锁定在2.2Ghz,没有重力感应，没有声音，帐篷模式也缺少驱动。