MrChromebox/scripts #100: Alternative boot menu not being displayed for Grunt / Careena

bobbarnes1981, 2019-11-29:
Hi,

I am using a HP Chromebook 14 (Careena / Grunt).

I used the firmware script to install RW_LEGACY and at first I thought it didn't work as CTRL-L just show a white screen with "Alternative Firmware Menu". However pressing 1 will launch Uboot and 2 will show the coreboot logo and allow access to the menu. Should a boot menu be displayed?

After choosing UEFI USB it does attempt to boot from a linux USB but the graphics are not in the right mode I think as it is sort or corrupted.

MrChromebox, 2019-11-29:
sounds like things are broken on Google's end unfortunately. You should be presented with a list of boot options when pressing CTRL+L (it works on other AMD Chromebooks). The display is corrupted (colors are off) and that's on the stock firmware as well. You can boot/run Ubuntu 19.04/19.10 without issue but some other distros seem to be having issues. Best you can do right now is file a bug report with Google

closed 2019-11-30 by bobbarnes1981

hackedXD, 2022-05-04:
Has this been fixed? I am struggling to open up my chromebook and so cant disable wp so help would be appreciated.

MrChromebox, 2022-05-04:
I don't know, I don't have one of these to test, and it's not something I can fix on my end. But either way, AMD Stoneyridge Chromebooks run Linux **terribly** so it's a waste of time either way