MrChromebox/scripts #101: Falco checksum fail

bambi666999, 2019-12-02:
I have been trying to flash full rom firmware on my HP Chromebook 14, and am stuck at the below error message.

....................................
Downloading Full ROM firmware
(coreboot_tiano-falco-mrchromebox_20191129.rom)

Firmware download checksum fail; download corrupted, cannot flash.
..................................

I followed the website's instruction and executed the below cmd line :

cd; curl -LO https://mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh

Thanks in advanced for your assistance!

bambi666999, 2019-12-02:
OK, I found the problem.  The file name is wrong in sources.sh.  It should be coreboot_tiano-falco-mrchromebox_20191201.rom.  So, I downloaded sources.sh manually, edited it, modified firmware-util.sh, and now I have full rom firmware!

Thank you for the great script/package!

closed 2019-12-02 by bambi666999

MrChromebox, 2019-12-02:
sorry about that, fixed