MrChromebox/scripts #104: Samsung Chromebook 3 graphics flashes while on "Please wait..." on Windows 10

Cyuubi, 2019-12-25:
I was trying to setup Windows 10, however I don't know if it's anything wrong with the EFI graphics implementation or something stupid. I'm unable to install Windows 10 simply due to the fact that "Please wait..." constantly hangs, it also flickers on and off.

Dunno if this is the right place to report this issue, I apologize if it's not. :)

MrChromebox, 2019-12-25:
The Samsung CB3 (CELES) is known to have unique issues running Windows 10.  Please see the Getting Started guide on reddit (https://www.reddit.com/r/chrultrabook/comments/aufp1q/getting_started_read_this_first/) which contains full instructions and a workaround for the issue you are having.

And for future reference, OS support questions are best handled there. I try to keep this issue tracker just for bugs/issues related to running the script itself

closed 2019-12-25 by MrChromebox

Cyuubi, 2019-12-25:
Yeah sorry about that, I found that just as you responded. It seems to be an issue with the HPET (High Precision Event Timer) implementation in the EFI firmware, I think.

MrChromebox, 2019-12-25:
I looked into that, and I'm not sure why the CB3 is problematic when no other device is. I even tried disabling the HPET with no improvement

JustinSUF, 2022-01-17:
I dunno if you guys are still alive but the workaround post was deleted.
does anyone have a archive of it or remember the solution?

MrChromebox, 2022-01-17:
@JustinSUF this isn't an issue with my script, and I don't provide Windows support. See the chrultrabook subreddit and DIscord channel for that

Jose1214pepe24, 2022-08-21:
I'm from Venezuela, I'm using google translate. to solve this problem, you have to use the windows installer, and once you have installed the os you have to reboot from usb to hit the repair computer button and start cmd (command prompt) and write this bcdedit /set {default} useplatformclock true bcdedit /set {default} useplatformclock true