MrChromebox/scripts #105: Lenovo ideapad 100s not recognized

1nikolas, 2019-12-27:
Hello. I have a Lenovo ideapad 100s (Windows version) and I'm trying to install Chrome OS Firmware to replace Windows with Chrome OS (if its even possible (??) ). I ran the script on bootable Ubuntu but cpu/device name is not recognized correctly. Also when I choose 9 nothing happens. Here is a screenshot 
[image]

Thanks in advance

MrChromebox, 2019-12-27:
You can't run my script, or use ChromeOS firmware, on a non-ChromeOS device. What you're doing won't work, since the hardware is different. I'd recommend looking at Project Croissant on XDA if you're trying to hack ChromeOS to run on standard PC hardware

closed 2019-12-27 by MrChromebox

1nikolas, 2019-12-27:
Thanks for your response. The problem is that I can't boot any chromium on this laptop even though this same model ships with chrome os.. I dont know whats wrong, it just stucks on the boot screen...

MrChromebox, 2019-12-27:
unfortunately that's not something I can help with, and even though Lenovo sells the same model with ChromeOS, the hardware is not exactly the same.