MrChromebox/scripts #106: trackpad not work,  eve update firmware in 2020

jingxi0109, 2020-01-11:
guys,
i use the last firmware in windows 10 1909 x64, tackpad not work, back to ChromeOS and Linux it works ok.

i tried to use the trackpad downgrade script always feedback not success, any suggestion about this ?

thanks~

MrChromebox, 2020-01-11:
well, clearly the problem is that you were not able to successfully downgrade the touchpad firmware -- the Windows drivers are not compatible with the newer version.  You need to do the following, in order, from ChromeOS:

* flash the UEFI firmware
* downgrade the touchpad firmware
* reboot

If the touchpad firmware is failing to downgrade from ChromeOS, then post the full, exact error from the script. Pic is fine.

jingxi0109, 2020-01-12:
downgrade the touchpad firmware in chromeOS response as below.
-------------------------------------------------------
Downloading touchpad firmware
(rose_v1.1.8546-ee1861e9e.bin)

Flashing touchpad firmware -- do not touch the touchpad while updating!

Error flashing touchpad firmware:

flashrom v0.9.9  : 980235d1 : Oct 24 2019 22:45:42 UTC on Linux 4.4.196-16940-geab5e6346fdb (x86_64)
flashrom was built with unknown PCI library, LLVM Clang 9.0.0 (/var/cache/chromeos-cache/distfiles/host/egit-src/llvm-project 6b043f051836635a1e88da4d0464e6569bd7b625), big endian
Command line (8 args): /usr/sbin/flashrom -p ec:type=tp -i EC_RW -w -o /tmp/flashrom.log rose_v1.1.8546-ee1861e9e.bin
Acquiring lock (timeout=180 sec)...
Opened file lock "/run/lock/firmware_utility_lock"
Lock acquired.
disable_power_management: Disabling power management.
Calibrating delay loop... OS timer resolution is 1 usecs, 2737M loops per second, 10 myus = 10 us, 100 myus = 98 us, 1000 myus = 982 us, 10000 myus = 10250 us, 4 myus = 4 us, OK.
Initializing internal programmer
Found candidate at: 00000500-00000528
Found coreboot table at 0x00000500.
Found candidate at: 00000000-00000538
Found coreboot table at 0x00000000.
coreboot table found at 0x7aabf000.
coreboot header(24) checksum: 9e08 table(1312) checksum: cafe entries: 43
Vendor ID: Google, part ID: Eve
DMI string system-manufacturer: "Google"

MrChromebox, 2020-01-13:
hmm, I don't see any error though

jingxi0109, 2020-01-13:
> hmm, I don't see any error though
here is the highlight error message.

### Error flashing touchpad firmware:

Downloading touchpad firmware
(rose_v1.1.8546-ee1861e9e.bin)

Flashing touchpad firmware -- do not touch the touchpad while updating!

### Error flashing touchpad firmware:

flashrom v0.9.9 : 980235d1 : Oct 24 2019 22:45:42 UTC on Linux 4.4.196-16940-geab5e6346fdb (x86_64)

MrChromebox, 2020-01-13:
that's the error printed from my script when the command doesn't return success, it then dumps the flashrom output recorded to the log file to provide detail. But there is no indication of what the error was in the flashrom log.

jingxi0109, 2020-01-17:
i follow the step and tried again from chromeOS. and with last uefi firmware, but the touchpad still not work in windows 10, any suggestion?

jingxi0109, 2020-01-17:
BTW, can the touchpad downgrade script work in GalliumOS?

MrChromebox, 2020-01-17:
the touchpad working or not working under Windows 10 is purely a function of the touchpad firmware version. If you're not able to downgrade the TP firmware, then nothing else matters; the UEFI firmware is not a factor at all. Your only other option is to use a ChromeOS recovery image from v74 or earlier, and not let it upgrade before you flash the UEFI firmware.

You can't downgrade the TP firmware from GalliumOS, only ChromeOS (or ChromiumOS).

MrChromebox, 2020-01-17:
just for kicks, I reset my PB to stock, updated to latest ChromeOS, and then flashed the UEFI firmware and downgraded the touchpad without issue. I'm not sure why you're having issues but it's certainly not a universal problem

jingxi0109, 2020-01-18:
thanks for your help. from your view where i need to check to find out the probelem is?

MrChromebox, 2020-01-18:
well, someone else reported that it didn't work under ChromeOS for them, but it did under a ubuntu live USB, maybe try that?

jingxi0109, 2020-01-30:
ChromeOS and Ubuntu live usb is working fine, just not work in windows.

MrChromebox, 2020-01-30:
@jingxi0109 you misunderstood. The touchpad will not work under Windows unless you downgrade the touchpad firmware, period. I suggested you do this via my script run from Ubuntu as someone reported successfully downgrading that way after not being able to from ChromeOS

jingxi0109, 2020-01-31:
i tried in ubuntu, the touchpad script not work as below.

Downloading touchpad firmware
(rose_v1.1.8546-ee1861e9e.bin)

Flashing touchpad firmware -- do not touch the touchpad while updating!

Error flashing touchpad firmware:

flashrom v0.9.9 chromium.googlesource.com/chromiumos/third_party/flashrom : c329008 : Oct 28 2017 12:29:02 UTC on Linux 5.0.0-23-generic (x86_64)
Calibrating delay loop... OK.
coreboot table found at 0x7aabb000.
WARNING: SPI Configuration Lockdown activated.
Error: Programmer initialization failed.

This function sometimes doesn't work under Linux, in which case it is
recommended to try under ChromiumOS.

jingxi0109, 2020-01-31:
any help?

 ChromeOS Firmware Utility Script [2020-01-17] 
 (c) Mr Chromebox <mrchromebox@gmail.com> 
*********************************************************
**   Device: Google Pixelbook (EVE)
** CPU Type: Intel KabyLake
**  Fw Type: Full ROM / UEFI (MrChromebox-4.11.1 01/10/2020)
**    Fw WP: Disabled
*********************************************************
**      1) Install/Update RW_LEGACY Firmware 
**      2) Install/Update BOOT_STUB Firmware 
** [WP] 3) Install/Update Full ROM Firmware 
** [WP] D) Downgrade Touchpad Firmware 
**      4) Set Boot Options (GBB flags)
**      5) Set Hardware ID (HWID) 
**      6) Remove ChromeOS Bitmaps 
**      7) Restore ChromeOS Bitmaps 
**      8) Restore Stock BOOT_STUB 
** [WP] 9) Restore Stock Firmware (full) 
*********************************************************
Select a menu option or

MrChromebox, 2020-01-31:
I don't have a good solution currently, since it works perfectly under ChromeOS/ChromiumOS for me. Did you try booting ChromiumOS (latest Arnoldthebat build) and downgrade from there?

jingxi0109, 2020-02-01:
under  chormeOS has failed, i will try ChromiumOS later, my chromebook eve is the i7 version is there any different with yours?

jingxi0109, 2020-02-03:
Below is feedback info under ChromiumOS,
Downloading touchpad firmware
(rose_v1.1.8546-ee1861e9e.bin)

Flashing touchpad firmware -- do not touch the touchpad while updating!

Error flashing touchpad firmware:

flashrom v0.9.9  : 7f47d094 : Jan 17 2020 10:03:03 UTC on Linux 4.14.167-15188-g9eb730d0560b (x86_64)
flashrom was built with unknown PCI library, LLVM Clang 10.0.0 (/var/cache/chromeos-cache/distfiles/host/egit-src/llvm-project 4e8231b5cf0f5f62c7a51a857e29f5be5cb55734), big endian
Command line (8 args): /usr/sbin/flashrom -p ec:type=tp -i EC_RW -w -o /tmp/flashrom.log rose_v1.1.8546-ee1861e9e.bin
Acquiring lock (timeout=180 sec)...
Opened file lock "/run/lock/firmware_utility_lock"
Lock acquired.
disable_power_management: Disabling power management.
Calibrating delay loop... OS timer resolution is 1 usecs, 1743M loops per second, 10 myus = 10 us, 100 myus = 97 us, 1000 myus = 1006 us, 10000 myus = 10360 us, 4 myus = 4 us, OK.
Initializing internal programmer
Found candidate at: 00000500-00000528
Found coreboot table at 0x00000500.
Found candidate at: 00000000-00000438
Found coreboot table at 0x00000000.
coreboot table found at 0x7aabb000.
coreboot header(24) checksum: a3fb table(1056) checksum: c613 entries: 35
Vendor ID: Google, part ID: Eve
Using External DMI decoder.
DMI string chassis-type: "Convertible"
DMI chassis-type is not specific enough.
DMI string system-manufacturer: "Google"
DMI string system-product-name: "Eve"
DMI string system-version: "1.0"
DMI string baseboard-manufacturer: "Google"
DMI string baseboard-product-name: "Eve"
DMI string baseboard-version: "1.0"
get_target_bus_from_chipset() returns 0x10.
Found chipset "Intel Skylake" with PCI ID 8086:9d24. Enabling flash write... 
BIOS_CNTL = 0x8b: BIOS Lock Enable: enabled, BIOS Write Enable: enabled
SPI BAR is = 0xfe010000
GCS = 0x8b: BIOS Interface Lock-Down: enabled, Boot BIOS Straps: 0x0 (SPI)
SPIBAR = 0x00007f4a1eaa5000 + 0x0000
ich_ generation 16
0x04: 0x3f00e800 (HSFSC)
WARNING: SPI Configuration Lockdown activated.
0x08: 0x00ffffc0 (FADDR)
0x50: 0x00004acb (FRAP)
BMWAG 0x00, BMRAG 0x00, BRWA 0x4a, BRRA 0xcb
0x54: 0x00000000 (FREG0: Flash Descriptor)
0x00000000-0x00000fff is read-only
0x58: 0x0fff0200 (FREG1: BIOS)
0x00200000-0x00ffffff is read-write
0x5C: 0x01ff0001 (FREG2: Management Engine)
0x00001000-0x001fffff is locked
0x60: 0x00007fff (FREG3: Gigabit Ethernet)
Gigabit Ethernet region is unused.
0x64: 0x00007fff (FREG4: Platform Data)
Platform Data region is unused.
0x68: 0x00007fff (FREG5: Device Expansion)
Device Expansion region is unused.
0x6C: 0x00007fff (FREG6: Reserved 1)
Reserved 1 region is unused.
0x70: 0x00007fff (FREG7: Reserved 2)
Reserved 2 region is unused.
0x74: 0x00007fff (FREG8: Embedded Controller)
Reading flash descriptors mapped by the chipset via FDOC/FDOD... done.
Embedded Controller region is unused.
0x84: 0x00000000 (PR0, unused)
0x88: 0x00000000 (PR1, unused)
0x8C: 0x00000000 (PR2, unused)
0x90: 0x00000000 (PR3, unused)
0x94: 0x00000000 (PR4, unused)
0x98: 0x00000000 (PR5, unused)
[... 104 lines cut ...]
CROS_EC is jumping to [EC_RO]
Aborting RWSIG jump.
CROS_EC jumped/rebooted to [EC_RO]
CROS_EC needs 2nd pass.
000000-0x7ffff:R fix_erasers_if_needed: kept all erasers
fill_sorted_erasers: using fallback eraser: region 2, function 0 total 0x60000 vs 0x7fdb8
040000..07ffff  20000 x 2 eraser 0
Erasing and writing flash chip... , 0x040000-0x05ffff: E__cros_ec_command_dev_v2(): Command 0x13 returned result: 4
__cros_ec_command_dev_v2(): Command 0x13 returned result: 4
__cros_ec_command_dev_v2(): Command 0x13 returned result: 4
 OLD[EC_RW] DENIED, 0x060000-0x07ffff: E__cros_ec_command_dev_v2(): Command 0x13 returned result: 4
__cros_ec_command_dev_v2(): Command 0x13 returned result: 4
__cros_ec_command_dev_v2(): Command 0x13 returned result: 4
 OLD[EC_RW] DENIED
SUCCESS.
RWSIG enabled: doing a cold reboot to enable WP.
Verifying flash... 000000..0x7ffff VERIFY FAILED at 0x00040014! Expected=0x1d, Read=0x4d, failed byte count from 0x00000000-0x0007ffff: 0x2
Your flash chip is in an unknown state.
Get help on IRC at irc.freenode.net (channel #flashrom) or
mail flashrom@flashrom.org with FAILED: your board name in the subject line!
-------------------------------------------------------------------------------
DO NOT REBOOT OR POWEROFF!
FAILED
Restoring PCI config space for 00:1f:5 reg 0xdc
restore_power_management: Re-enabling power management.

MrChromebox, 2020-02-03:
@jingxi0109 what does the following report when run under ChromiumOS?

`sudo flashrom -p ec --wp-status`  
`sudo flashrom -p ec:type=tp --wp-status`  
`sudo crossystem wpsw_cur`

jingxi0109, 2020-02-05:
below is the result under ChromeOS, the last line maybe "sudo crossystem wpsw_curl"

-------
chronos@localhost / $ sudo flashrom -p ec --wp-status

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

flashrom v0.9.9  : 7f47d094 : Jan 17 2020 10:03:03 UTC on Linux 4.4.210-17188-g2a4d401f342c (x86_64)
flashrom v0.9.9  : 7f47d094 : Jan 17 2020 10:03:03 UTC on Linux 4.4.210-17188-g2a4d401f342c (x86_64)
coreboot table found at 0x7aabf000.
Found chipset "Intel Skylake". Enabling flash write... WARNING: SPI Configuration Lockdown activated.
OK.
WP: status: 0x80
WP: status.srp0: 1
WP: write protect is enabled.
WP: write protect range: start=0x00000000, len=0x00040000
------
chronos@localhost / $ sudo flashrom -p ec:type=tp --wp-status
flashrom v0.9.9  : 7f47d094 : Jan 17 2020 10:03:03 UTC on Linux 4.4.210-17188-g2a4d401f342c (x86_64)
flashrom v0.9.9  : 7f47d094 : Jan 17 2020 10:03:03 UTC on Linux 4.4.210-17188-g2a4d401f342c (x86_64)
coreboot table found at 0x7aabf000.
Found chipset "Intel Skylake". Enabling flash write... WARNING: SPI Configuration Lockdown activated.
OK.
WP: status: 0x80
WP: status.srp0: 1
WP: write protect is enabled.
WP: write protect range: start=0x00000000, len=0x00040000

-----
1chronos@localhost / $sudo crossystem wpsw_curl
Invalid parameter name: wpsw_curl

Usage:
  crossystem [--all]
    Prints all parameters with descriptions and current values.
    If --all is specified, prints even normally hidden fields.
  crossystem [param1 [param2 [...]]]
    Prints the current value(s) of the parameter(s).
  crossystem [param1=value1] [param2=value2 [...]]]
    Sets the parameter(s) to the specified value(s).
  crossystem [param1?value1] [param2?value2 [...]]]
    Checks if the parameter(s) all contain the specified value(s).
Stops at the first error.
Valid parameters:
  arch                     [RO/str] Platform architecture
  backup_nvram_request     [RW/int] Backup the nvram somewhere at the next boot. Cleared on success.
  battery_cutoff_request   [RW/int] Cut off battery and shutdown on next boot
  block_devmode            [RW/int] Block all use of developer mode
  clear_tpm_owner_done     [RW/int] Clear TPM owner done
  clear_tpm_owner_request  [RW/int] Clear TPM owner on next boot
  cros_debug               [RO/int] OS should allow debug features
  dbg_reset                [RW/int] Debug reset mode request
  debug_build              [RO/int] OS image built for debug features
  dev_boot_legacy          [RW/int] Enable developer mode boot Legacy OSes
  dev_boot_signed_only     [RW/int] Enable developer mode boot only from official kernels
  dev_boot_usb             [RW/int] Enable developer mode boot from USB/SD
  dev_default_boot         [RW/str] Default boot from disk, legacy or usb
  dev_enable_udc           [RW/int] Enable USB Device Controller
  devsw_boot               [RO/int] Developer switch position at boot
  devsw_cur                [RO/int] Developer switch current position
  diagnostic_request       [RW/int] Request diagnostic rom run on next boot
  disable_dev_request      [RW/int] Disable virtual dev-mode on next boot
  ecfw_act                 [RO/str] Active EC firmware
  post_ec_sync_delay       [RW/int] Short delay after EC software sync (persistent, writable, eve only)
  fmap_base                [RO/int] Main firmware flashmap physical address
[... 16 lines cut ...]
  mainfw_act               [RO/str] Active main firmware
  mainfw_type              [RO/str] Active main firmware type
  nvram_cleared            [RW/int] Have NV settings been lost?  Write 0 to clear
  display_request          [RW/int] Should we initialize the display at boot?
  phase_enforcement        [RO/int] Board should have full security settings applied
  recovery_reason          [RO/int] Recovery mode reason for current boot
  recovery_request         [RW/int] Recovery mode request
  recovery_subcode         [RW/int] Recovery reason subcode
  recoverysw_boot          [RO/int] Recovery switch position at boot
  recoverysw_cur           [RO/int] Recovery switch current position
  recoverysw_ec_boot       [RO/int] Recovery switch position at EC boot
  recoverysw_is_virtual    [RO/int] Recovery switch is virtual
  ro_fwid                  [RO/str] Read-only firmware ID
  tpm_attack               [RW/int] TPM was interrupted since this flag was cleared
  tpm_fwver                [RO/int] Firmware version stored in TPM
  tpm_kernver              [RO/int] Kernel version stored in TPM
  tpm_rebooted             [RO/int] TPM requesting repeated reboot (vboot2)
  tried_fwb                [RO/int] Tried firmware B before A this boot
  try_ro_sync              [RO/int] try read only software sync
  vdat_flags               [RO/int] Flags from VbSharedData
  vdat_lfdebug             [RO/str] LoadFirmware() debug data (not in print-all)
  vdat_lkdebug             [RO/str] LoadKernel() debug data (not in print-all)
  wipeout_request          [RW/int] Firmware requested factory reset (wipeout)
  wpsw_boot                [RO/int] Firmware write protect hardware switch position at boot
  wpsw_cur                 [RO/int] Firmware write protect hardware switch current position

MrChromebox, 2020-02-05:
> below is the result under ChromeOS, the last line maybe "sudo crossystem wpsw_curl"

cur (short for current), not curl. the output to `sudo crossystem wpsw_cur` is either 0 or 1

MrChromebox, 2020-02-05:
also, try:  
`sudo flashrom -p ec --wp-disable`
`sudo flashrom -p ec --wp-range 0 0`

if those succeed, run my script and try downgrading the TP firmware again

jingxi0109, 2020-02-06:
> > below is the result under ChromeOS, the last line maybe "sudo crossystem wpsw_curl"
> 
> cur (short for current), not curl. the output to `sudo crossystem wpsw_cur` is either 0 or 1

return 1;

jingxi0109, 2020-02-06:
1chronos@localhost / $ sudo flashrom -p ec --wp-disable
flashrom v0.9.9  : 7f47d094 : Jan 17 2020 10:03:03 UTC on Linux 4.4.210-17188-g2a4d401f342c (x86_64)
flashrom v0.9.9  : 7f47d094 : Jan 17 2020 10:03:03 UTC on Linux 4.4.210-17188-g2a4d401f342c (x86_64)
coreboot table found at 0x7aabf000.
Found chipset "Intel Skylake". Enabling flash write... WARNING: SPI Configuration Lockdown activated.
OK.
FAILED: RO_AT_BOOT is not clear.
FAILED
chronos@localhost / $

jingxi0109, 2020-02-06:
chronos@localhost / $ sudo flashrom -p ec --wp-range 0 0
flashrom v0.9.9  : 7f47d094 : Jan 17 2020 10:03:03 UTC on Linux 4.4.210-17188-g2a4d401f342c (x86_64)
flashrom v0.9.9  : 7f47d094 : Jan 17 2020 10:03:03 UTC on Linux 4.4.210-17188-g2a4d401f342c (x86_64)
coreboot table found at 0x7aabf000.
Found chipset "Intel Skylake". Enabling flash write... WARNING: SPI Configuration Lockdown activated.
OK.
FAILED: RO_AT_BOOT is not clear.
FAILED

jingxi0109, 2020-02-06:
------
Downloading touchpad firmware
(rose_v1.1.8546-ee1861e9e.bin)

Flashing touchpad firmware -- do not touch the touchpad while updating!

Error flashing touchpad firmware:

flashrom v0.9.9  : 7f47d094 : Jan 17 2020 10:03:03 UTC on Linux 4.4.210-17188-g2a4d401f342c (x86_64)
flashrom was built with unknown PCI library, LLVM Clang 10.0.0 (/var/cache/chromeos-cache/distfiles/host/egit-src/llvm-project 4e8231b5cf0f5f62c7a51a857e29f5be5cb55734), big endian
Command line (8 args): /usr/sbin/flashrom -p ec:type=tp -i EC_RW -w -o /tmp/flashrom.log rose_v1.1.8546-ee1861e9e.bin
Acquiring lock (timeout=180 sec)...
Opened file lock "/run/lock/firmware_utility_lock"
Lock acquired.
disable_power_management: Disabling power management.
Calibrating delay loop... OS timer resolution is 1 usecs, 2674M loops per second, 10 myus = 10 us, 100 myus = 98 us, 1000 myus = 988 us, 10000 myus = 9873 us, 4 myus = 4 us, OK.
Initializing internal programmer
Found candidate at: 00000500-00000528
Found coreboot table at 0x00000500.
Found candidate at: 00000000-00000560
Found coreboot table at 0x00000000.
coreboot table found at 0x7aabf000.
coreboot header(24) checksum: 1b0d table(1352) checksum: 4dd0 entries: 45
Vendor ID: Google, part ID: Eve
Using External DMI decoder.
DMI string chassis-type: "Desktop"
DMI string system-manufacturer: "Google"
DMI string system-product-name: "Eve"
DMI string system-version: "1.0"
DMI string baseboard-manufacturer: "Google"
DMI string baseboard-product-name: "Eve"
DMI string baseboard-version: "1.0"
get_target_bus_from_chipset() returns 0x10.
Found chipset "Intel Skylake" with PCI ID 8086:9d24. Enabling flash write... 
BIOS_CNTL = 0x8b: BIOS Lock Enable: enabled, BIOS Write Enable: enabled
SPI BAR is = 0xd123d000
GCS = 0x8b: BIOS Interface Lock-Down: enabled, Boot BIOS Straps: 0x0 (SPI)
SPIBAR = 0x00007d28db961000 + 0x0000
ich_ generation 16
0x04: 0x0010e000 (HSFSC)
WARNING: SPI Configuration Lockdown activated.
0x08: 0x00000000 (FADDR)
0x50: 0x00004acb (FRAP)
BMWAG 0x00, BMRAG 0x00, BRWA 0x4a, BRRA 0xcb
0x54: 0x00000000 (FREG0: Flash Descriptor)
0x00000000-0x00000fff is read-only
0x58: 0x0fff0200 (FREG1: BIOS)
0x00200000-0x00ffffff is read-write
0x5C: 0x01ff0001 (FREG2: Management Engine)
0x00001000-0x001fffff is locked
0x60: 0x00007fff (FREG3: Gigabit Ethernet)
Gigabit Ethernet region is unused.
0x64: 0x00007fff (FREG4: Platform Data)
Platform Data region is unused.
0x68: 0x00007fff (FREG5: Device Expansion)
Device Expansion region is unused.
0x6C: 0x00007fff (FREG6: Reserved 1)
Reserved 1 region is unused.
0x70: 0x00007fff (FREG7: Reserved 2)
Reserved 2 region is unused.
0x74: 0x00007fff (FREG8: Embedded Controller)
Reading flash descriptors mapped by the chipset via FDOC/FDOD... done.
Embedded Controller region is unused.
0x84: 0x00000000 (PR0, unused)
0x88: 0x00000000 (PR1, unused)
0x8C: 0x00000000 (PR2, unused)
0x90: 0x00000000 (PR3, unused)
0x94: 0x00000000 (PR4, unused)
0x98: 0x00000000 (PR5, unused)
0x9C: 0x00000000 (PR6, unused)
[... 110 lines cut ...]
fill_sorted_erasers: using fallback eraser: region 2, function 0 total 0x60000 vs 0x7fdb8
040000..07ffff  20000 x 2 eraser 0
Erasing and writing flash chip... , 0x040000-0x05ffff: E__cros_ec_command_dev_v2(): Command 0x13 returned result: 4
__cros_ec_command_dev_v2(): Command 0x13 returned result: 4
__cros_ec_command_dev_v2(): Command 0x13 returned result: 4
 OLD[EC_RW] DENIED, 0x060000-0x07ffff: E__cros_ec_command_dev_v2(): Command 0x13 returned result: 4
__cros_ec_command_dev_v2(): Command 0x13 returned result: 4
__cros_ec_command_dev_v2(): Command 0x13 returned result: 4
 OLD[EC_RW] DENIED
SUCCESS.
RWSIG enabled: doing a cold reboot to enable WP.
Verifying flash... 000000..0x7ffff VERIFY FAILED at 0x00040014! Expected=0x1d, Read=0x4d, failed byte count from 0x00000000-0x0007ffff: 0x2
Your flash chip is in an unknown state.
Get help on IRC at irc.freenode.net (channel #flashrom) or
mail flashrom@flashrom.org with FAILED: your board name in the subject line!
-------------------------------------------------------------------------------
DO NOT REBOOT OR POWEROFF!
FAILED
Restoring PCI config space for 00:1f:5 reg 0xdc
restore_power_management: Re-enabling power management.

This function sometimes doesn't work under Linux, in which case it is
recommended to try under ChromiumOS.

Press [Enter] to return to the main menu.

MrChromebox, 2020-02-06:
> return 1;

so you still have WP enabled, which means you didn't follow the instructions to properly disable it. This is likely the cause of all your issues. I don't know why you tried to downgrade again when the commands clearly said failed.

Follow the instructions at https://wiki.mrchromebox.tech/Firmware_Write_Protect and verify that the firmware WP is disabled like it tells you to before trying again =/

closed 2020-02-06 by MrChromebox